trezor/trezor-firmware #1: Rethink PassphraseStateRequest

slush0, 2019-04-15:
Make the workflow same for client for both non-passphrase device and passphrase device. Currently, client receives state only when passphrase is enabled.

prusnak, 2019-05-17:
Related to Protocol redesign: https://github.com/trezor/trezor-firmware/issues/79

andrewkozlik, 2019-05-17:
Some good points about passphrase design were made here: https://www.reddit.com/r/Bitcoin/comments/bp0nqs/trezor_or_ledger_which_one_is_better/enof517?utm_source=share&utm_medium=web2x

To summarize:
- Make it possible to easily switch between passphrases on the fly.
- Get rid of enable/disable passphrase as a configuration option. There should be no indication of whether in the past the device was used with a (non-empty) passphrase or not.
- Use a 2nd PIN to unlock the device and automatically enter a preconfigured passphrase.

I think the first two points can be solved by adding a "Switch passphrase" button to the web wallet. As for the third point, I actually think that this is in contradiction with plausible deniability. For one thing, there is likely to be some side-channel information, which could reveal whether the primary or secondary PIN was entered. So in general I am not too keen on supporting multiple PINs. However, I would like to make it possible to configure a default passphrase which would be automatically used upon start-up. The reason is that people may want to use a passphrase for various reasons:
- Plausible deniability, i.e. to have a real wallet and a decoy wallet.
- Providing additional protection, because they do not trust the PIN and hardware to protect their seed if the wallet is stolen.
- Providing additional protection in case their paper backup is stolen.
Say you are worried about protecting your paper backup but you trust the hardware and PIN to provide sufficient protection. In that case entering a passphrase during every start-up can be a real burden, which is why you'd like to configure it.

prusnak, 2019-05-17:
I strongly agree with the first and second Reddit suggestions. Strongly Disagree with the third one.

>  I would like to make it possible to configure a default passphrase which would be automatically used upon start-up.

Where would this passphrase be stored? I think it is antipattern to store it anywhere. For various usability and security reasons.

matejcik, 2019-05-17:
agree with @prusnak. Mainly because if passphrase is stored, it becomes another piece of "static" data that you don't remember after a month of not needing it, and you need to back it up somehow. Either you add it to the seed (which defeats point (3)), or you handle it separately (which gives you a brand new point of failure).

andrewkozlik, 2019-05-17:
> Where would this passphrase be stored? I think it is antipattern to store it anywhere. For various usability and security reasons.

It would be stored encrypted in the storage just like the mnemonic. This doesn't prevent you from using more passphrases. For example, you can have:
- passphrase1 to access your regular accounts which you use often and where you have say around $1000, and
- passphrase2 to access your savings accounts which you use rarely and where you have larger sums.

You configure passphrase1 to be stored on Trezor and used as default, whereas passphrase2 is not stored anywhere. That way you can easily access your regular accounts. If somebody steals your paper backup, they are not able to access neither your regular accounts nor your savings accounts. If they steal your Trezor and guess the PIN or extract the sensitive data by some means, then they gain access to your regular accounts but not to your savings accounts. So this feature merely provides a way to fine-tune your risk/usability tradeoff.

prusnak, 2019-05-17:
I don't like that, honestly. It introduces confusion as people will surely forget the stored passphrase.

tsusanka, 2019-06-21:
Consider splitting up reset device and backup device as part of this issue, because it is a breaking change for wallet / third parties as well.

nitram147, 2019-07-08:
Long time ago I was thinking about opening an issue "[feature] Change or validate passphrase".
I finally found time and I found that similar issue (this) is already open. So I want to share with you my opinion.

Each time when I use my trezor (model T) I find two things very inconvenient.
1.) When I'm using multiple passphrases I have to replug my device each time when I want to change it. It's very annoying and also I don't think it's great for the trezor's usb-c connector.
2.) When I use new passphrase (without any balance) I'm always not sure whether or not I made a typo.
I'm using long passphrases so I can verify the beginning of passphrase only when I start to type it. Later I see only "<" which unconsciously makes me  paranoid.

I propose a solution as implementing new "passphrase menu". You can access it for example by long press on homescreen for xy seconds.

In this menu you can do three things:
1.) Exit the menu
2.) Change passphrase (on the fly as mentioned in upper comments)
3.) Validate current passphrase.

Validation can be implemented in 2 ways - secure or less secure one.

In case of secure one, you will have to enter passphrase once more and trezor will let you know whether passphrases match. (Same way as recovery seed validation works).

Less secure way is that you can just show your current passphrase.

When somebody has access to unlocked trezor he can of course steal your funds so it "doesn't matter" whether or not he can read current passphrase. But I see another dangerous scenario. You and your family/friends share same seed but use different passphrases. Your friend find your unlocked trezor, read the passphrase and note it to a paper to misuse it later.

Maybe the best solution would be to first validate passphrase and in case they match show passphrase on the display. In this case you can prevent doing same typo twice and thinking that you entered correct passphrase. When you show passphrase on the display (as in case when you show any text on trezor) you can clearly read whole passphrase as opposed to case when you are entering long passphrase and cannot see the beginning.

fresheneesz, 2019-10-13:
> if passphrase is stored, it becomes another piece of "static" data that you don't remember after month of not needing it

> It introduces confusion and people will surely forget the stored passphrase.

This doesn't have to be the case. The device can store a passphrase temporarily - for example up to 4 hours. So if you're using your wallet a lot in a day, you only put in the passphrase once. But after 4 hours, the device deletes the passphrase (next time its on probably, but would be better if it could delete it even if its not plugged in somehow). If this is the case, they'd have to use their passphrase regularly and would remember it. 

> Where would this passphrase be stored?

> Either you add it to the seed .., or you handle it separately (which gives you a brand new point of failure).

This I don't know. If the physical security of the Trezor were more reliable, you could store it with the seed and it might be fine.  Especially if the device could manage to ensure the passphrase is only stored ephemerally (eg 4 hours). The problem with requiring a passphrase every time is it will make people sloppy, and a video of someone typing in their passphrase is all that's required to compromise it, so limiting how often the passphrase is necessary could improve security. But storing the passphrase also means all the data necessary to compromise your wallet exists somewhere in the physical device and could be extracted - at least until the passphrase is deleted.

tsusanka, 2019-10-23:
Outcome from today's meeting:

- `pass_protection` flag will remain, but suite will always set it to true.

The protocol will be as follows:

```
-> Initialize (including state, we will remove `use_empty_passphrase`)
<- Features
-> PingState/GetState/AssureState (name TBD)
    if pass_protection == true:
        <- PassphraseRequest (with the `on_device` flag removed)
        ... Client will prompt the user to enter the passphrase, there will be an option to input the passphrase on the device
        -> PassphraseAck (text = "")  - (new field `on_device` or we might set text to None, TBD)
<- SuccessState (state = abcd) - name TDB
```

[Screenshot from the whiteboard](https://user-images.githubusercontent.com/1835345/67405313-dfd97680-f5b4-11e9-8aa2-c102a22f1812.jpg)

tsusanka, 2019-11-11:
@prusnak what was the point of https://github.com/trezor/trezor-firmware/commit/4097f254d537e659e94d1c6d5764f414d39509c5? We will clear the cache including passphrase in the new design, correct?

prusnak, 2019-11-11:
> @prusnak what was the point of 4097f25?

This was to fix the suboptimal UX because everytime when the web wallet forgot the passphrase, user had to select on device where they want to enter it.

closed 2020-02-11 by tsusanka

matejcik, 2020-02-11:
[image]

:100: :100: :100: :fireworks:

tsusanka, 2020-02-17:
## Testing

This was already partially tested but let's add some test scenarios so we are on the same side:

- Test the usual passphrase workflow against the develop branch of Suite:
  - Start a wallet without passphrase.
  - Start multiple wallets with passphrase.
  - Disable passphrase.
  - Send some coins a to a passphrase wallet, unplug and check.
  - Any other passphrase-related actions.
- Test the usual passphrase workflow against Wallet. **Note: ** the work is not yet done there.
- Make sure that you are asked only once for each passphrase. As long as you work with the same passphrase, the passphrase should not get requested again.

As mentioned QA tested this already, so let's wait for https://github.com/trezor/trezor-firmware/issues/839 and test this again together with new feature.

tsusanka, 2020-03-30:
One QA scenario I have forgotten:

- Enable the "passphrase always on device" setting using `trezorctl set passphrase enabled -f`.
- Test against both Wallet and Suite that the passphrase is always requested on the device.

sorooris, 2020-04-07:
1. QA OK

2.3.0 (0b7a8449f)
Suite: bc1225231da00906c40a46d8cf5eac32bdfda206
Wallet (stage): 59e011aabb0d494ba569028e3eb8a2ad51c9eabc
Next Wallet (stage): 08826e1e8555a83d84eeedb55b5995c508ed8518

Passphrase was correctly requested only on the device, per session.

2. QA OK

2.3.0 (0b7a8449f)
1.9.0 (0b7a8449f)
Electrum 3.3.8

Passphrase was correctly prompted only once per session, tested send & receive, sign & verify.

A trivial visual bug was present in the passphrase entry prompt in the app.
[image]

tsusanka, 2020-04-07:
The visual bug should be fixed in Electrum's master, se https://github.com/spesmilo/electrum/pull/6064#issuecomment-607450626 . So I consider this done :tada: