trezor/trezor-firmware #11: bootloader: jump_to_firmware can lead to undefined behavior

mcudev, 2019-04-15:
currently the `jump_to_firmware` function can lead to undefined behavior if an exception is taken after `SCB_VTOR` is set and before the firmware's `reset_handler` has completed. what i mean is that the firmware exception handlers are written in C and the sections that get initialized in SRAM are put there by the `reset_handler`. this is an informational issue right now because i don't see this being exploitable with the firmware's current set of exception handlers. the model t already considered this possibility and handles it differently.

to see this in action in a dev board, set a break point at https://github.com/trezor/trezor-mcu/blob/master/util.h#L65 and touch the reset pin on your board to cause clock instability which will trigger the `nmi_handler` for the clock security system. since the firmware's vectors are now in use, when you step in the debugger, you'll be transported to the exception handler at the firmware's flash memory address. thus, we're running firmware code before completing the bootloader and the subsequent firmware initialization.

EDIT: adding a note that the model t also clears out the registers just in-case anything sensitive was in them. reference https://github.com/trezor/trezor-core/blob/master/embed/trezorhal/util.s

tsusanka, 2020-06-05:
@prusnak is this still up to date? Is it something we would like to include in the next bootloader release?

prusnak, 2020-06-05:
@tsusanka Let's have @hiviah assess the issue.

mcudev, 2021-01-16:
Perhaps the most that is needed is to add a comment that warns of this possibility in the legacy T1 code where this informational issue could apply: `trezor-firmware/legacy/util.h` function `jump_to_firmware`.

The TT code already avoids this possible undefined behavior, so I'm not worried about it there.
Other than that, this can probably be closed. At the time that I thought of this less than obvious execution possibility, opening an issue was just the way to communicate it that came to mind.
Mitigating steps already in-place to reduce probability of this becoming a problem include:
1. `jump_to_firmware` is already called after the unofficial firmware user warning
1. `jump_to_firmware` already transitions control to untrusted firmware after mpu config
1. unless i missed it, i don't see the `SCB_VTOR` being changed from the bootloader's handlers for untrusted firmware. this would limit potential impacts to only officially signed firmware.
1. sram is zeroed before calling `jump_to_firmware`
1. i haven't seen any existing handlers in the officially signed firmware use data sections that may not be initialized yet (which is the biggest concern that can lead to undefined behavior). handlers don't change often either. so, it's mainly a don't shoot-yourself-in-the-foot kind of thing, i think.

mcudev, 2021-04-15:
closing as not actionable

closed 2021-04-15 by mcudev