trezor/trezor-firmware #12: [meta] Future improvements of the boot process

prusnak, 2019-04-15:
This is an issue to discuss future improvements of the boot process.

- [ ] update the firmware so it works with the original VTOR table from the bootloader
- [ ] update the firmware so it works in unprivileged mode, most probably enabling the supervisor calls will do the trick: https://github.com/trezor/trezor-mcu/pull/323
- [ ] create a mechanism so SIGNED firmware can indicate the bootloader it needs to use its own VTOR table and/or it needs privileged mode

tsusanka, 2020-06-05:
@prusnak is this still up to date? Is it something we would like to include in the next bootloader release?

tsusanka, 2020-06-05:
I am assigning @hiviah as well to assess this.

matejcik, 2021-10-13:
alright, thanks.
pinging @prusnak in case we want to return to this, but for now I'm closing the issue.

closed 2021-10-13 by matejcik