trezor/trezor-firmware #14: Investigate dropping required fields

prusnak, 2019-04-15:
and replacing them with optional.

We need to asses the impact on Trezor code and other parts (wallet, connect, etc.)

effort leading to upgrading to protobuf 3: https://github.com/trezor/trezor-common/issues/228

prusnak, 2019-04-15:
this is a pre-requisite for https://github.com/trezor/trezor-firmware/issues/16

matejcik, 2019-08-02:
If i understand protobuf correctly, we are not using the required/optional dichotomy for anything anyway. Changing `required` to `optional` in the defs is for free. The only thing that `required` does is failing to parse a message that does not contain that field.

That means: if our code sends the required field, it will be in the message, so non-updated parsers will work fine.
If our code does NOT send the field, non-updated parsers are failing right now anyway.

prusnak, 2019-08-02:
Yes, all code needs to correctly handle the default values in case the field is not present.

ZdenekSL, 2019-08-22:
Closed because of #379

closed 2019-08-22 by ZdenekSL