trezor/trezor-firmware #16: Investigate upgrading to Protobuf v3

jpochyla, 2019-04-15:
https://developers.google.com/protocol-buffers/docs/proto3

The only possible problem now seems to be the default values. Cases where the explicit (v2) default value does not match the implicit (zero) values from v3:

https://github.com/trezor/trezor-common/blob/7d618036ab50504e245ad8651c4558d413596228/protob/messages-bitcoin.proto#L128

- Is `version == 0` a valid value?

https://github.com/trezor/trezor-common/blob/7d618036ab50504e245ad8651c4558d413596228/protob/messages-bitcoin.proto#L207

- Is `sequence == 0` a value value?

If we decide we need to keep the compatibility with v2, there is one trick we can use: https://stackoverflow.com/a/33204511

prusnak, 2019-04-15:
https://github.com/trezor/trezor-firmware/issues/14 is a pre-requisite
\

matejcik, 2019-08-02:
what are our reasons for upgrading to proto3?

There are only minor changes in the wire format, and the semantic changes are discussed in #379. Am I missing something more to this?

prusnak, 2019-08-02:
I guess there is no reason to upgrade except for the feeling that proto2 might get obsoleted by the tooling in the future and we will need to upgrade anyway. If we think this not happen I agree that we might stick to proto2 and maybe even reintroduce `required` fields as needed.

ZdenekSL, 2019-08-26:
Closed in favor of #379

closed 2019-08-26 by ZdenekSL