trezor/trezor-firmware #4: Create a new special intermediary firmware

tsusanka, 2019-04-15:
To make onboarding easier let's create a new firmware, which will update bootloader to its newest version and firmware to the newest version, so user does not have to update (and unplug) its device several times.

tsusanka, 2019-04-29:
@prusnak this is the issue we've discussed at standup, right? I'm moving it to 06/2019 then. cc @mroz22

tsusanka, 2019-10-23:
We will create a new "artificial" firmware, which will perform the first upgrade and then the second one right away. Of course Wallet/Suite needs to support this.

We might also do the soft resets while we are at it.

mroz22, 2019-12-05:
Just a little irritating question on possibly delivery of this feature @ZdenekSL 
I am asking to know whether I should put more effort into workarounds on trezor-suite side or not.

tsusanka, 2020-05-25:
Does it still make sense if we are doing _port_?

tsusanka, 2020-06-05:
We should do this as part of #24. 

What we need is:
- Firmware that updates bootloaders on older T1 devices.
- This is important for new devices UX:
  - Host firstly sends this intermediary firmware, which updates the bootloader to its latest version (since the device does not contain a firmware no confirmation is needed)
  - Host then sends the actual up to date firmware (since the device does not[1] contain a firmware no confirmation is needed)
  - In the end, only one replugging should be needed in the whole onboarding process (that one after uploading the intermediary firmware).

[1] Technically, the device already contains firmware (that intermediary one), so there must be some logic how to solve this. Maybe the intermediary firmware will erase itself after it reflashes the bootloader?

tsusanka, 2020-06-05:
Although this is not a bootloader issue per se I am assigning the labels because it is bootloader-related and we want to do it as part of some other bootloader tasks.

hiviah, 2020-06-18:
So to recap the strategy from intermediate FW's point of view:

1. it updates bootloader to latest version that is packed together with FW
1. copies small shim into RAM to allow current firmware to be erased (compiled to fixed address or position independent)
1. jumps to shim
1. shim deletes firmware
1. shim soft-resets device

tsusanka, 2020-08-28:
This bundles the bootloader.dat file. That is the bootloader latest 1.8.0 version right? Could we build a new version 1.8.1 from master and include that one? It includes some new features such us the soft-reset so we should update to the latest IMHO. Is there any reason not to?

hiviah, 2020-08-31:
It's the exactly same version that is used for normal firmware (it's a symlink). So it's either 1.8.0 or 1.8.1.

closed 2020-09-11 by tsusanka

tsusanka, 2020-09-25:
Let's test this when it is in Suite, that means after #2338.