trezor/trezor-firmware #22: Keyboard still get gets stucked after "heavier" testing

tsusanka, 2019-04-16:
This is somewhat connected to https://github.com/trezor/trezor-core/issues/509 . Although we have improved the behaviour it is still possible to get the trezor completely stucked. Steps to reproduce:

- do a recovery (or anything that leads to a keyboard on Trezor)
- press any button with letters, hold it and swipe to another button while holding
- sometimes the device gets stucked

I've built debug firmware and there is no error in the log. Strangely enough, I am sometimes able to do some other operations.

pankrats, 2019-05-28:
It still happens with new firmware 2.1.1. during Recovery (and dry-run recovery). 

Also, with fw 2.1.1 touch screen **still stays** irresponsive even after web wallet reload. 

1. Do a dry-run recovery, device gets stuck at some point: 
[image]

2. Refresh the web wallet (F5), result: 
[image]

matejcik, 2019-05-28:
@pankrats can you reproduce the freeze doing anything other than recovery? typing a long passphrase maybe?

pankrats, 2019-05-28:
@matejcik yes, it happens during the passphrase typing as well! 
but it's harder to reproduce.
[image]
[image]

The result after web wallet refresh is still the same (screen is irresponsive).

matejcik, 2019-05-28:
it's almost certainly #144 then :/

GlitchXploitr, 2019-05-29:
Is there an eta for a firmware update that will fix this issue? I just got my model T today and decided to dry test my seed before moving any funds to my trezor. Had the frozen screen issue on the first dry test while entering the 12th word. Firmware version 2.1.0

ZdenekSL, 2019-06-26:
Please, retest with new firmware 2.1.2

ZdenekSL, 2019-08-22:
Please, retest with new firmware 2.1.4

bosomt, 2019-08-25:
@ZdenekSL it's still in 2.1.5 fw version ;)

tsusanka, 2019-08-26:
@bosomt how "hard" was it to reproduce this?

bosomt, 2019-08-26:
@tsusanka i can always reproduce it via device passphrase enter dialogue
device stars to respond after wallet is refreshed

matejcik, 2019-08-26:
@bosomt please also test if you can reproduce the freeze when entering recovery words.

If not, let's close this issue and open a new one that is specific to passphrase entry. This particular issue has a history that is too deep and convoluted.

Also please test if this happens with a very long passphrase, or with short passphrase but large number of key presses (i.e., enter some letters, erase them, enter again, etc.)

bosomt, 2019-08-27:
@matejcik 
1) i think we can close it but maybe @pankrats wants to say something about it

2) it's not related to length, it freezes even with very shot passphrase,

3) i tried many recoveries of shamir seed so lots of typing and no freeze since last fix

matejcik, 2019-08-27:
OK, closing in favor of (very nice) #444

closed 2019-08-27 by matejcik