trezor/trezor-firmware #24: [meta] port codebase to Trezor One hardware

prusnak, 2019-04-16:
- [x] HAL
  - [x] display
  - [x] touch
  - [x] usb
- [x] modules
  - [x] modtrezorconfig
  - [x] modtrezorcrypto
  - [x] modtrezorio
  - [x] modtrezorui
  - [x] modtrezorutils
- [ ] boardloader/bootloader
- [ ] application code

tsusanka, 2019-08-22:
Doing https://github.com/trezor/trezor-firmware/issues/435 beforehand would help us a lot.

tsusanka, 2020-07-02:
We have agreed on with Product that we would like to keep the PIN entry on T1 as it is, that means using the pin matrix. Yes this has a drawback that the USB needs to be turned on beforehand, however we do not want introduce a new feature into the Port process.

prusnak, 2020-07-02:
> We have agreed on with Product that we would like to keep the PIN entry on T1 as it is, that means using the pin matrix. Yes this has a drawback that the USB needs to be turned on beforehand, however we do not want introduce a new feature into the Port process.

This makes FIDO2 PIN entry on device impossible. We could theoretically allow entering the FIDO PIN on host (FIDO allows us to specify this) - but this is a bad idea when the FIDO PIN will be the same as a device PIN (current situation) - introducing different PIN for FIDO also seems like a UX hell.

closed 2024-09-24 by Hannsek