trezor/trezor-firmware #33: IDEA: Implement TOTP/HOTP

prusnak, 2019-04-16:
It's pretty straightforward to implement TOTP/HOTP in the following way:

- we store the TOTP/HOTP secret on the computer (Password manager maybe?)
- we extend CipherKeyValue or SignIdentity message to provide extra functionality which takes this secret (and counter for HOTP or timestamp for TOTP), encrypts that secret with a private key derived using a path
- numeric code is shown on the device display, optionally it can be returned back to the computer

matejcik, 2019-05-27:
it would be pretty cool if Trezor could *type in* the TOTP code, acting as a USB keyboard

if we did have that functionality, Password Manager could make use of it too. not sure if it's something we want though

andrewkozlik, 2019-06-06:
> it would be pretty cool if Trezor could _type in_ the TOTP code, acting as a USB keyboard
> 
> if we did have that functionality, Password Manager could make use of it too. not sure if it's something we want though

I had the same idea for Password Manager, because I am not at all happy with how it works. What I got stuck on is the fact that a keyboard does not send characters to the computer, but scan codes which are mapped to characters depending on the system keyboard layout. This means that we would need to know the layout to correctly type in the password. As far as I was able to find, there is no way to get or set the system keyboard layout via USB. The HID descriptor has a bCountryCode field, but it's probably not much use. Here is what the HID spec says about it:

> The value bCountryCode identifies which country the hardware is localized for. Most hardware is not localized and thus this value would be zero (0). However, keyboards may use the field to indicate the language of the key caps. Devices are not required to place a value other than zero in this field, but some operating environments may require this information.

So we would either have to rely on the user to use the same layout every time or get the layout via some process running on the host, like Trezor Bridge.

However, the TOTP/HOTP codes are numeric, so we could use the numeric keypad scan codes, which do not get remapped in different keyboard layouts.

andrewkozlik, 2019-06-06:
Just as I thought, we are not the first to come up with this idea: https://onlykey.io/ The way it works is you tell the device which keyboard layout to use.
Source: https://github.com/trustcrypto/OnlyKey-Firmware
Docs: https://docs.crp.to/usersguide.html

AlexITC, 2019-06-06:
I have been thinking about this idea for some time, and I would love to have it available on trezor.

The one time codes are usually 6 digits only, is there really a need to work like a keyboard? I'd be happy to just type them. In my opinion, a bigger problem is how to recover the seeds.

On recovering the seeds, I see two possible options:
- Backing up them to a file encrypted by a password generated by the master seed.
- Proposing to the apps that trezor generates the TOPT seed (which could be difficult to get sites adopting this), this way, we could generate the seed in a deterministic way based on the website url.

StoneMoe, 2019-09-09:
Idea: 
It will be awesome if the password manager database can be stored in sd card, and Trezor acts as keyboard to input the password with just one "Confirm" click!

MarkusZoppelt, 2019-11-16:
Strongly support this!

Storing the encrypted database on an SD card as an alternative to Dropbox or Google Drive is much more compliant to corporate settings. I know many european companies where employees are not even allowed to sign in to US cloud providers at their workstations.

Plus, what if I want to store my Dropbox password on TPM as well? As of right now, I need that somewhere else as I need to sign into Dropbox first before I can use TPM.

Maybe open a new issue for TPM SD card storage?

replaysMike, 2020-02-17:
I'd love to see this as well

prusnak, 2020-02-18:
Our current stance is that we should create more pressure on websites to implement FIDO2 as soon as possible and drop TOTP/HOTP because they are obsolete methods.

StoneMoe, 2020-02-19:
> Our current stance is that we should all create more pressure on websites to implement FIDO2 as soon as possible and drop TOTP/HOTP because they are obsolete methods.

That's absolutly right to keep moving forward on FIDO2 but it's necessary to support TOTP/HOTP for backward compatible IMO (even disable it by default)

zsoltsandor, 2020-05-13:
I second this, a lot of websites still only support xOTP.

AlexITC, 2020-05-13:
Getting back to this, the feature should be simple now that trezor seems to support an sd card.

zsoltsandor, 2020-05-13:
It should be available for Trezor One as well, if possible.

brianddk, 2020-05-20:
I would love to imagine that other websites would move to FIDO2, but U2F has been out for over 5 years and hardly any sites have picked it up.  Many sites still consider SMS as good 2FA. 

+1 for TOTP on SD

heavypackets, 2020-06-10:
The CTO of SatoshiLabs has recently said, bluntly, that this isn't a feature they will ever support. 

https://old.reddit.com/r/TREZOR/comments/gclfkm/using_for_2fa/fpcaf52/ 

There was an attempted PR of this functionality in the password manager last year but it was rejected for very opaque reasons: https://github.com/trezor/trezor-password-manager/pull/68

Lack of TOTP is the reason why I must stick with Yubikey. I have too many work-related and financial apps that only support TOTP. A handful of nerds has, obviously, not put noticed pressure on companies to move away from SMS, let alone TOTP. This position on TOTP is nearing zealotry at this point, considering how many personal security postures it would immediately improve.

Yubikey + Yubikey Authenticator is a fair alternative: https://github.com/Yubico/yubioath-desktop

rikur, 2021-05-18:
Would love to see this, please consider it again. I would love to have them stored securely in Trezor and have them backed up as part of my seed.

indolering, 2021-06-02:
At risk of further spamming this thread ... a UX engineer should be in charge of this decision, [not the CTO](https://old.reddit.com/r/TREZOR/comments/gclfkm/using_for_2fa/fpcaf52/).  While I applaud the moral stance, I sadly doubt that WebAuth is going to ever see widespread support.  I know of one major US bank who supports anything other than SMS multi-factor. 

The main problem is **that most U2F/WebAuth security tokens can't be restored from offline backups**.  Imagine being a business faced the prospect of angry customers who lost their Yubikey and didn't bother registering multiple backups.  I know I don't have time to register 3 security tokens for the dozens of services I use.  The entire reason Authy and Duo exist is because they can handle device transitions even if the customer loses their equipment.

Trezor doesn't have the weight to push a better solution.  From a usability perspective, you should be addressing your **customers** very real need to secure access to TOTP codes in a way that gets backed-up.  Rejecting patches from community members just leaves us reliant on crappy phone apps and provides an opening for your competitors.

sunknudsen, 2022-08-05:
@prusnak

> Our current stance is that we should all create more pressure on websites to implement FIDO2 as soon as possible and drop TOTP/HOTP because they are obsolete methods.

I agree FIDO2 is more secure than TOTP but, at least in the context of YubiKeys, FIDO2 backups (precisely the lack thereof) are a huge sovereignty issue.

Simply put, YubiKey’s implementation does not allow backups serving enterprise environments where IT departments can recover accounts.

For personal environments, I believe TOTP is the only option one has to securely (yet inconveniently) backup hashes.