trezor/trezor-firmware #34: update micropython dependency to 1.10

prusnak, 2019-04-16:
- [ ] Review mpconfig.h and figure out if we can disable some optional features.
- [ ] Can we enable `MICROPY_OPT_CACHE_MAP_LOOKUP_IN_BYTECODE`?
- [ ] Can we use newly-introduced `MP_STATIC_ASSERT` somewhere?
- [ ] Can we somehow benefit from pystack?
- [ ] our uctypes.sizeof patch probably no longer necessary - see micropython/micropython#3101 + micropython/micropython@2411f42
- [ ] remove SOF callback from the CDC code - see micropython/micropython@0f6f86c

jpochyla, 2019-07-29:
https://github.com/micropython/micropython/commit/fa07deda9f129c1992726d2a51c6eaac168247f0

tsusanka, 2020-01-14:
Done by https://github.com/trezor/trezor-firmware/pull/789

closed 2020-01-14 by tsusanka