trezor/trezor-firmware #36: Bootloader mode - text overflow on vendor change screen

sorooris, 2019-04-16:
Trezor T

Starting the T2 device in bootloader mode and installing new firmware displays a Vendor change screen where the text overflows to the right.
[image]

tsusanka, 2020-02-17:
We will solve this by https://github.com/trezor/trezor-firmware/issues/659

ondracja, 2022-05-18:
I'm reopenning the issue, it seems that it have slipped back as an regression.

prusnak, 2022-05-18:
> I'm reopenning the issue, it seems that it have slipped back as an regression.

No, it hasn't. The bootloader with the fix has not been released yet.

hynek-jina, 2022-08-04:
Bootloader will be redesigned

closed 2022-08-04 by hynek-jina