trezor/trezor-firmware #37: Add support for coinjoin / Wasabi

slush0, 2019-04-16:
To support Trezor in coinjoin wallets like Wasabi (https://github.com/zkSNACKs/Meta/issues/30), some basic functionality for auto-signing / pre-confirmed signing needs to be implemented in Trezor Core.

Let's discuss how this can be possible in secure way and eventually implement it as standalone firmware app.

prusnak, 2019-10-10:
Let's refactor the signing code first: https://github.com/trezor/trezor-firmware/issues/617

andrewkozlik, 2020-05-22:
I wrote up a spec for proofs of ownership ([SLIP-0019](https://github.com/satoshilabs/slips/blob/slips-19-20-coinjoin-proofs/slip-0019.md)), which is currently under [review](https://github.com/satoshilabs/slips/pull/936). However, I have been thinking that for some applications we might not even need proofs of ownership. The idea that I am contemplating is this:

If Trezor is asked to sign a transaction with external inputs, all of which have already been signed, then it can safely sign the remaining inputs to the transaction without risking the kind of attack described [here](https://lists.linuxfoundation.org/pipermail/bitcoin-dev/2017-August/014843.html) and [here](https://github.com/satoshilabs/slips/blob/slips-19-20-coinjoin-proofs/slip-0019.md#motivation).

Of course for this to work, Trezor needs to be able to verify the signatures of all the external inputs against their scriptPubKeys, so the implementation still involves quite a bit of work in Trezor. The major advantage to this solution is that the other parties providing inputs don't need to support SLIP-0019 and there is no need to exchange the proofs. In particular this could be used in certain PayJoin applications like [BIP-0079](https://github.com/bitcoin/bips/blob/master/bip-0079.mediawiki) (BustaPay). However, I wonder if this idea isn't overlooking some vulnerability...

Hannsek, 2023-01-24:
We do not need to have this epic anymore. Therefore I am closing it now.

closed 2023-01-24 by Hannsek