trezor/trezor-firmware #38: Support 'EXTERNAL' for TransactionInput.ScriptType

RHavar, 2019-04-16:
The firmware doesn't appear to support 'EXTERNAL'   for TransactionInput.ScriptType, which is essential for coinjoins. I'm trying to build a simple bip79 front-end for trezor, but this appears to make it impossible.

Comment from issue: https://github.com/trezor/connect/issues/247

prusnak wrote:
> If you want to try implementing it, you should add another case when txi.script_type == InputScriptType.EXTERNAL here:
> https://github.com/trezor/trezor-core/blob/b9926a9fffe8a593beb3adb79e5fc429233e41a3/src/apps/wallet/sign_tx/signing.py#L148

prusnak, 2019-10-10:
Let's refactor the signing code first: https://github.com/trezor/trezor-firmware/issues/617

tsusanka, 2020-06-09:
Implementing SLIP-19 is needed for this. Moved to #1052.

closed 2020-06-09 by tsusanka