trezor/trezor-firmware #39: Implement full WebAuthn

prusnak, 2019-04-16:
Currently, we do U2F, but let's expand the functionality to full WebAuthn support.

prusnak, 2019-05-23:
Mainly we are interested in parsing Application+User data from the protocol and showing that on the display.

andrewkozlik, 2019-05-31:
Pull request: https://github.com/trezor/trezor-firmware/pull/204

Regression testing: Focus on U2F and Cardano. Ensure that logging in with a U2F credential that was created prior to firmware upgrade works correctly.

andrewkozlik, 2019-07-09:
Subtasks:
#245 Add support for FIDO2 resident credentials P3 Medium S4 Low core enhancement
#206 Implement FIDO2 credential selection P3 Medium S4 Low core enhancement ui
#205 Redesign FIDO2 credential ID format P3 Medium S4 Low core enhancement
#31 Modify U2F/webauthn app to use Keychain

andrewkozlik, 2019-08-13:
The FIDO2 implementation for Trezor T is currently undergoing review https://github.com/trezor/trezor-firmware/pull/394. If all goes well, we can release this in September. There are several options for the release:
1. Release FIDO2 in stable firmware for all users.
2. Release FIDO2 for beta testing in stable firmware. The feature would need to be activated via trezorctl.
3. Release FIDO2 for beta testing in separate firmware.

I would prefer option 3 as it is the safest and cleanest solution. The full release of FIDO2 can happen in October. From the marketing perspective there is no rush, because in September the focus will be on Shamir Backup. I expect @prusnak to make the final decision on this and @ZdenekSL to communicate the decision with marketing.

tsusanka, 2019-09-18:
The only thing missing is #245, which is tracked in its own issue. Closing.

closed 2019-09-18 by tsusanka

Quiark, 2019-12-07:
Is there any plan to implement FIDO2 for Trezor One?

On the shop page it's marked as supported ...

[image]

prusnak, 2019-12-07:
@Quiark That's a bug on our website which will be fixed soon. Currrently, we don't plan to support FIDO2 on Trezor One.

1000i100, 2020-05-18:
What a shame !
Is there lot's of work to do it, or is it marketing choice ?