trezor/trezor-firmware #49: add support for PSBT (BIP-174)

prusnak, 2019-04-17:


willcl-ark, 2019-04-25:
If implemented, would this support Multisig PSBTs?

prusnak, 2019-04-25:
Yes

prusnak, 2019-10-10:
Let's refactor the signing code first: https://github.com/trezor/trezor-firmware/issues/617

urza, 2020-09-11:
Hello, may I ask what is current status of PSBT support in Trezor?

I would appreciate if someone could provide overview, something like:

- Which devices support PSBT (T / One)
- What of BIP174 functionality is supported and if there are any limitations, what isn't possible.
- How can PSBT be handled from developer's perspective (integration into other software - which tools/library/methods to use and ideally short example)
- How can PSBT be handled from user's perspective (does web wallet support it? electrum? core with hwi?)

:pray:

prusnak, 2020-09-11:
> Which devices support PSBT (T / One)

Right now, you can sign PSBT via [HWI](https://github.com/bitcoin-core/HWI) with both Trezor One and Model T. We will add PSBT signing via SD-card, but there are things with more priority (native hardware Coinjoin, etc.) in our list first.

matejcik, 2020-09-11:
> what is current status of PSBT support in Trezor?

There is no direct support of the PSBT encoding in hardware.

(I believe this issue was originally motivated by storing PSBT files on the SD card for model T, which is not done at the moment.)

Client software that can decode the PSBT format can translate it to Trezor calls, so any transaction that Trezor can sign, it can in theory do so via PSBT.

That is, you can generate a PSBT for offline signing, transfer it to an offline host on a flash drive or something, use Electrum or HWI with connected Trezor to sign it, and then transfer it back for broadcast.

There's limited support for multisig, because ... well (_vague gestures_). There are no good standards around multisig UX. BIP-174 allows encoding (most of?) the necessary raw data, but that doesn't translate to good support.

Multi-party signing is now possible on Trezor T via `EXTERNAL` script type, with either a pre-signed input (supported in BIP-174) or SLIP-19 ownership proofs (not supported, could be added via a vendor extension). AFAIK neither Electrum nor HWI supports that yet.

Arbitrary scripts are not supported on Trezor, only the standard ones.

> * How can PSBT be handled from developer's perspective

HWI, which can be used in Python as `hwilib`

> * How can PSBT be handled from user's perspective

HWI or Electrum

jimmysong, 2020-09-30:
So is there an ETA for PSBT support via SD card? I would really like this for my multisig setup.

fooedge2, 2021-04-13:
Any update on BIP174 with sd card?

prusnak, 2021-04-13:
PSBTs via SD card are not on the immediate roadmap. You can sign PSBTs via HWI with no issues.