trezor/trezor-firmware #54: Device freezes when displaying Bitcoin Cash Receiving address if I lock the computer screen

pankrats, 2019-04-17:
Chrome, Linux
wallet.trezor.io

How to reproduce: 
1. Switch to Bitcoin Cash
2. Go to Receive tab 
3. Click on Show full address, wait for the address to appear on Trezor Screen
4. Lock your computer (for 1 min approx)
5. Restore your computer. 

Result: the screen becomes irresponsive. 

[image]

bosomt, 2019-05-27:
i was not able to reproduce it with latest 2.1.1
I think it was already fixed.
Please consider to closing this issue ;)

pankrats, 2019-05-27:
It still happens. Please, keep the issue opened. 

Firmware 2.1.1.
Chrome, OS X

[image]

ZdenekSL, 2019-06-25:
@pankrats Please retest with a new firmware.

pankrats, 2019-06-25:
It still happens with 2.1.2 (core-2019-07-1ec5d0). 
important note: it works perfectly with WebUSB, the bug happens only when using **Bridge**

[image]

prusnak, 2019-06-25:
> It works perfectly with WebUSB, the bug happens only when using Bridge

Sounds like a wallet issue then.