trezor/trezor-firmware #55: boardloader-bootloader-firmware reverse crosscheck

prusnak, 2019-04-17:
T1 firmware checks bootloader hash against the known whitelist, see https://github.com/trezor/trezor-mcu/blob/9afaeff36ff891ab9dbe48b0f514a273975f8cd0/firmware/bl_check.c#L29-L46

Maybe we should do the same for T2, that is:
* bootloader checks the boardloader hash against the whitelist
* firmware checks the bootloader hash against the whitelist

tsusanka, 2019-11-07:
> firmware checks the bootloader hash against the whitelist

@prusnak believes this is already done

matejcik, 2024-08-22:
> * firmware checks the bootloader hash against the whitelist

currently, firmware checks the bootloader against the bundled version, and overwrites if there is a mismatch
this will break if the bootloader flipped the monotonic downgrade record, but that can't be avoided, because it's the already-released versions that will do this

i believe that this mechanism is generally fine for bootloader-firmware cross-check?

prusnak, 2024-08-22:
> i believe that this mechanism is generally fine for bootloader-firmware cross-check?

I guess so.

The thing that remains is bootloader checking the boardloader.

andrewkozlik, 2024-08-26:
> currently, firmware checks the bootloader against the bundled version, and overwrites if there is a mismatch

Since 115f7daaaa the firmware does not downgrade the bootloader if the monotonic version counter of the installed bootloader is higher. It's a good thing, because it doesn't brick the device when you downgrade firmware too much. It's a bad thing, because a fake bootloader can just claim a high `current_bld_hdr->monotonic` value and the firmware won't overwrite it. So AFAICS it doesn't work as a bootloader-firmware cross-check.

It should be easy to get this working safely by calling `error_shutdown()` here:
https://github.com/trezor/trezor-firmware/blob/c8b811bac51c835545c5f52ec4873c3f0e4e3385/core/embed/firmware/bl_check.c#L201-L204
That way the firmware ensures one of the following states:
- The bootloader is already the same as the bundled version. Firmware runs.
- The bootloader is downgraded or upgraded to the bundled version.  Firmware runs.
- The bootloader should be downgraded to the bundled version, but can't due to the monotonic flag. Firmware doesn't run.

If I am not mistaken, then this implies that the firmware only runs on a device whose bootloader is the same as the bundled version, i.e. an authentic bootloader.

To be on the safe side, the firmware should probably force a reboot after the new bootloader is written.