trezor/trezor-firmware #58: Emphasize coin in each screen

jpochyla, 2019-04-17:


ZdenekSL, 2019-06-25:
Probrat s Petrem jak předejít zvolení špatného coinu díky podobným zkratnám (asi ikona).

mmahut, 2019-06-25:
Please keep the public conversation in English.

ZdenekSL, 2019-06-25:
We need to discuss with Peter how to prevent selecting the wrong coin with a similar shortcut (probably an icon).

ZdenekSL, 2019-10-24:
Part of #53

tsusanka, 2020-08-03:
Since we are currently limiting derivation paths to each coin this is less important than it used to be. The design and Suite should also help with this significantly. Therefore we have decided in the Product team we do not want to implement this at the moment.

matejcik, 2020-08-04:
> Since we are currently limiting derivation paths to each coin this is less important than it used to be.

That's a weird rationale.
The original issue is that the user can be confused/tricked into signing a transaction of the wrong funds on the wrong chain. So this very much still exists regardless of derivation paths. This is purely a UX issue, Suite can't help because it is on the host side, so "design" is the only place to resolve this, and I think that this issue is specifically about "keep this thing in mind for a redesign".

tsusanka, 2020-08-04:
> The original issue is that the user can be confused/tricked into signing a transaction of the wrong funds on the wrong chain.

Our rational was that the user can no longer be tricked because when they are on coin X they can't use paths for Y. But you raising a point when user would like to use X coin with X paths but is tricked to use Y coin with Y paths, do I understand it correctly?

Our rational was that in transaction signing Trezor displays the amount with the coin abbreviation so user should notice if they are signing ETH instead of BTC. But you are right that the difference between BTC and BTG for example is very small. And in message signing Trezor enforces the prefix so that should be fine as well. But the first point is something we should reconsider.

matejcik, 2020-08-04:
> user would like to use X coin with X paths but is tricked to use Y coin with Y paths, do I understand it correctly?
(...)
> difference between BTC and BTG for example is very small

Exactly. Plus, this is even more relevant in case of tokens.

tsusanka, 2021-02-19:
> I think that this issue is specifically about "keep this thing in mind for a redesign".

We have discussed this with @matejzak with respect to the new redesign and he was against it. Mainly due to the maintenance reasons mentioning it was not that easy to make the icons nice and clean in Suite. He was strongly advocating for Trezor's design to be clear and simple. So I am suggesting to close this one.

tsusanka, 2021-10-05:
Please reopen if the rationale above does not seem sufficient.

closed 2021-10-05 by tsusanka