trezor/trezor-firmware #59: Implement Shamir secret sharing SLIP-0039

slush0, 2019-04-17:
Allow creating recovery seeds using Shamir secret sharing, and allow device recovery from such seeds.

https://github.com/satoshilabs/slips/blob/master/slip-0039.md

tsusanka, 2019-04-26:
Some work is done in the tsusanka/slip39 branch. In particular:

- **core:** common/mnemonic.py was refactored to introduce slip39
- **core:** slip39 functions in storage (although we may want to refactor this as we discussed)
- **core:** basic workflow
- **common:** messages
- **python:** added `-x` to `trezorctl reset-device` to init device using SLIP-39

The new asignee should try out:
- trezorctl recovery-device
- trezorctl reset-device -x

tsusanka, 2019-06-06:
Note: 

1) We need to choose BIP-39/SLIP-39 during device init. Not during backup.
Although we could create BIP-39 at start, work with that and then during backup ask (and then convert BIP-39 to SLIP-39 if applicable), the main problem is passphrase, which works differently in each standard.

2) We need to think through, if the current process when the backup fails completely on interruption is reasonable. The backup process will now take significantly longer

prusnak, 2019-06-06:
> We need to choose BIP-39/SLIP-39 during device init.

Sure we'd need shamir flag in the `ResetDevice` message, not in the `BackupDevice` message

ZdenekSL, 2019-07-02:
TBD in #269

closed 2019-07-02 by ZdenekSL