trezor/trezor-firmware #60: bootloader: Add "Wipe device" button

slush0, 2019-04-17:
Allow to wipe device without connecting to the computer.

Related to #301, where USB is disabled in bootloader, and complement to on-device management #302 in cases PIN is not known.

tsusanka, 2020-06-05:
If we decide to do #1039 this is not needed.

tsusanka, 2020-06-05:
Product wants to do this. I am removing also "UX needed", because if we wish to do some more UX/design modifications we should go with #1049.

tsusanka, 2020-06-09:
Let's discuss where exactly the Wipe Button should be. 

[image]

tsusanka, 2020-06-15:
Product has decided we want to do this in #1049.

Hannsek, 2023-02-15:
Until now, if you stole the trezor, you couldn't use it. With this, you can wipe the stolen trezor and start using it. Am I right?

prusnak, 2023-02-15:
> Until now, if you stole the trezor, you couldn't use it. With this, you can wipe the stolen trezor and start using it. Am I right?

No. You can still run `trezorctl device wipe` when the device is in the bootloader mode to wipe the device without knowing the PIN.

Hannsek, 2023-02-15:
ok, thanks. will be solved by this: 
[image]