trezor/trezor-firmware #63: Support for Erase PIN

slush0, 2019-04-17:
Allow setting of special PIN, which will immediately erase Trezor storage, ideally returning device to factory defaults.

Expected changes:
a) New enum PinType with choices STANDARD, ERASE
b) New flag 'type' in ChangePin, using PinType
c) Check in ChangePin, that both PINs are different
d) In Pin dialog in boot time, if entered PIN == Erase PIN, drop device's storage

There are various use cases for such feature, including tricks like hand-writen PIN nearby Trezor, so random thief will erase the device himself.

andrewkozlik, 2019-12-04:
Done, https://github.com/trezor/trezor-firmware/pull/720.

closed 2019-12-04 by andrewkozlik

andrewkozlik, 2019-12-16:
@trezor/qa 
- This feature is implemented for both the T1 and the TT.
- You'll need the latest trezorctl from the master branch (get it using `git` and run it via `pipenv`).
- To set the wipe code use: `trezorctl set wipe-code`
- To remove the wipe code use: `trezorctl set wipe-code -r`
- Check that if the user is prompted to enter their current PIN and they enter the wipe code instead, it causes the device to get wiped immediately.
- Check that it's not possible to set the wipe code to the same value as the PIN.
- Check that it's not possible to set the PIN to the same value as the wipe code.
- Check that removing the wipe code disables the feature correctly.
- Check that it's not possible to tell whether the device has a wipe code set (`trezorctl get-features`) when the device is locked.

sorooris, 2019-12-30:
@andrewkozlik 
- This feature is implemented for both the T1 and the TT.
do we have a build for T1 available?

prusnak, 2020-01-01:
@sorooris not this time