trezor/trezor-firmware #64: bootloader: Upload firmware over SD card

slush0, 2019-04-17:
Allow updating of both bootloader and firmware over SD card. Currently, only bootloader can be updated this way.

tsusanka, 2020-06-05:
We could do this as part of BL 1.9 but it is only nice to have feature and we should not slow down 1.9 because of this.

tsusanka, 2020-06-09:
We can do this but it would have to be in two steps:

- Install bootloader via SD card using boardloader (we have that already).
- Install firmware via SD card using bootloader (we can do that).

We can't however make this one-step - that would require boardloader change.

tsusanka, 2020-06-09:
We believe having filesystem on the SD card is kind of a prerequisite for this. Otherwise users would have to `dd` the firmwares making this feature even more nerdy.

tsusanka, 2020-06-15:
We have discussed this in a Product meeting and some good arguments were brought up.

First, is filesystem really a prerequisite? We were brainstorming that it is:
1. A nerdy feature and using some programs to "burn" ISO installation programs to USB/whatever is not that rare.
2. Having filesystem can bring other problems. Did user formatted the card to FAT? Not NTFS or something else? It seems like this is not the easiest way either.
3. It bloats the bootloader obviously by adding the filesystem.

So in a nutshell Product is OK with going the binary way - users will have to use `dd` or some similar tools.

---

Also Product is okay with having this in two steps. A good point was raised that after we release this and it gets included in the newly manufactured Trezor Ts at some point this will be possible right away with just one step.

J-esusleal, 2022-01-10:
> Allow updating of both bootloader and firmware over SD card. Currently, only bootloader can be updated this way.