trezor/trezor-firmware #66: Confidential Assets tracking issue

apoelstra, 2019-04-17:
It would be great to have support for Confidental Assets (CA), which is used in the Elements blockchain platform and in Liquid, the commercial sidechain based on Elements. I'm happy to help move this forward, and am available by email at apoelstra at blockstream.com for any clarification or guidance.

CA is described in this blog post: https://blockstream.com/2017/04/03/blockstream-releases-elements-confidential-assets.html
It is essentially the same as Confidential Transactions, which involves the use of Pedersen commitments in place of amounts, alongside rangeproofs, which I believe is already supported as part of Trezor's Monero support. CA mades one of the two EC generators variable and adds a new "asset surjection proof" which is described here: https://github.com/ElementsProject/secp256k1-zkp/blob/secp256k1-zkp/src/modules/surjection/surjection.md

We've implemented CA in full as part of libsecp256k1-zkp, our fork of the Bitcoin Core libsecp256k1 library.

For referenc, the API for making surjection proofs is here: https://github.com/ElementsProject/secp256k1-zkp/blob/secp256k1-zkp/include/secp256k1_surjectionproof.h
and the core crypto functionality is here:
https://github.com/ElementsProject/secp256k1-zkp/blob/secp256k1-zkp/src/modules/surjection/main_impl.h
https://github.com/ElementsProject/secp256k1-zkp/blob/secp256k1-zkp/src/modules/surjection/surjection_impl.h

Cheers

real-or-random, 2019-04-18:
My plan is to run the libsecp256k1-zkp test on the device to make sure everything works alright and maybe to provide some benchmarks. The current test suite is rather large (binary size) and probably needs a lot of memory. So I'll probably need to build a firmware just with the testsuite. Do you have any suggestions how to do that easily and where to start? Or maybe an entirely different approach?

romanz, 2019-05-28:
An update about our progress:

In the last few months, we have been working on a Confidential Assets [proof-of-concept](https://github.com/romanz/trezor-firmware/commits/liquid) support for TREZOR and we would like to start the review & upstreaming process for this feature.
I would be happy to change and adapt the code according to your feedback.

I plan to start with (relatively) small PRs for the parts of the code that are (relatively) easy to test and do not require significant changes in the existing code (e.g. the additional cryptographic APIs, confidential address generation and confidential output unblinding).