trezor/trezor-firmware #69: Better display ethereum function calls

ligi, 2019-04-17:
Currently when calling ethereum functions you only see the raw hex data - also often you do not see all of it as the hex gets long fast and I was not able to scroll.
The hex can be easily made shorter and more digestible by humans by resolving the function from the 4byte signature - e.g. via https://github.com/ethereum-lists/4bytes and then displaying the function name with parameters. So a really long hex string that might not fit on the trezor and is hard to parse by a human can easily fit on the display of the trezor and better digested by humans.
A problem might be the size of the signature database - I see 2 solutions here:
 - leverage the SD-Card
 - reduce the database to commonly used signatures only - we can parse all transactions currently done on chain - sort the 4bytes by usage and define a cut-off point

matejcik, 2020-06-22:
linking to #15

Hannsek, 2023-03-17:
duplicate? #2874

matejcik, 2023-03-17:
yes, this has more info so i suggest closing #2874 and keeping this one

Hannsek, 2023-03-17:
_Copied from #2874:_

Every transaction in Ethereum can carry additional input data (e.g. ```0x095ea7b30000000000000000000000008bc3702c35d33e5df7cb0f06cb72a0c34ae0c56f00000000000000000000000000000000000000000000000ee5c13efe85190000```) (https://etherscan.io/tx/0x2b930225479934eda949c3c2b0f3af5d5fd60136f7c9f0d5bbabf569def1f8a8). The first 4 bytes of this input data, i.e. `0x095ea7b3`, specify which function in the smart contract gets executed. These 4 bytes are not human readable. Therefore, there exists signature databases which help to map the hash (4 bytes) to human-readable form.

**Signature databases:**
- [https://www.4byte.directory/signatures/](https://www.4byte.directory/signatures/)
- [https://github.com/volsa/etherface](https://github.com/volsa/etherface)
- (https://www.etherface.io/statistics)

**[How to read data](https://towardsdatascience.com/decoding-ethereum-smart-contract-data-eed513a65f76)**

**example**
```
0x38ed1739000000000000000000000000000000000000000000000000000000009502f900000000000000000000000000000000000000000000a07e38bf71936cbe39594100000000000000000000000000000000000000000000000000000000000000a00000000000000000000000003c02cebb49f6e8f1fc96158099ffa064bbfee38b00000000000000000000000000000000000000000000000000000000616e11230000000000000000000000000000000000000000000000000000000000000003000000000000000000000000a0b86991c6218b36c1d19d4a2e9eb0ce3606eb48000000000000000000000000c02aaa39b223fe8d0a0e5c4f27ead9083c756cc2000000000000000000000000528b3e98c63ce21c6f680b713918e0f89dfae555
```
- `0x` -  indicator that the hexadecimal
- `38ed1739` - hashed signature of the function being called (every 2 hex characters represent a byte)
- The rest of the bytes are hashes of the arguments being passed to the function

_decoded data_:
```
function called:  swapExactTokensForTokens
arguments:  {
  "amountIn": 2500000000,
  "amountOutMin": 194024196127819599854524737,
  "path": [
    "0xA0b86991c6218b36c1d19D4a2e9Eb0cE3606eB48",
    "0xC02aaA39b223FE8D0A0e5C4F27eAD9083C756Cc2",
    "0x528B3e98c63cE21C6f680b713918E0F89DfaE555"
  ],
  "to": "0x3c02cebB49F6e8f1FC96158099fFA064bBfeE38B",
  "deadline": 1634603299
}
```

Possible next steps: 
- preload the well known smart contract [addresses](https://docs.uniswap.org/contracts/v3/reference/deployments) → label it → show the label next to the address (otherwise label it as _unknown_)
- label the Smart Contract address user is authorising and highlight an unknown address to give an alert
- address after parameter “to” (most of the time same as address of the sender) - compare “to” address with “from” address → if not same → warning