trezor/trezor-firmware #73: Display dry run result on device display

michal-novacek, 2019-04-17:
The device responds with a success or failure message after a `recovery_device` with `dry_run: true` is completed. It would be nice if it also displayed the result of the dry run on display.

This is mostly for consistency with T1 and web wallet. The risks of not displaying the result on device are minimal, since modifying it doesn't really benefit potential attacker.

pankrats, 2019-06-13:
There is the same issue in trezor-qa folder : https://github.com/satoshilabs/trezor-qa/issues/34

pankrats, 2019-06-13:
Expected: The result of "Check recovery seed" procedure should be displayed **on the device screen** (and preferably on the device screen only). 

At the moment, with Trezor T (firmware 2.1.1) the result is displayed in the wallet only. Nothing is displayed on device screen.
[image]

Trezor One (fw 1.8.1) displays the result on the device screen and in the wallet as well. 

[image]
[image]

Reported by @sipak

tsusanka, 2019-07-29:
This was done during the works on Shamir and the persistance.

closed 2019-07-29 by tsusanka