trezor/trezor-firmware #75: Configurable auto-lock feature

slush0, 2019-04-17:
Trezor One have constant-time timeout for locking device, which may be too long for some and frustrating for heavy users who works in trusted environment.

Model T does not have auto-locking yet. Let's make the timeout configurable over ApplySettings() with the possibility to turn the feature off completely.

Also, the wake event should be a human interaction (touch event), not wire communication, to protect device from misbehaving app on desktop.

tsusanka, 2019-04-25:
Let's do this in the following milestone, because it should solve some user-unfriendliness in Wallet. In particular, it prompts the user for passphrase entry selection (host vs device) way too often.

On 'clear cache' we should not forget the passphrase, but just lock the device. The passphrase will remain in cache.

prusnak, 2019-05-13:
Not erasing passphrase on ClearSession implemented in 4097f254d537e659e94d1c6d5764f414d39509c5

matejcik, 2019-05-13:
what's the purpose of ClearSession if it doesn't clear the session?
i mean, what is wallet using it for?

prusnak, 2019-05-13:
According to how it is being used now the message should have been called `LockDevice`.

matejcik, 2020-04-14:
#949 is a necessary prerequisite of this. After that is done, hardness of auto-locking is decreased, so I updated the estimate.

brianddk, 2020-04-16:
> what's the purpose of ClearSession if it doesn't clear the session?

@matejcik , so the `trezorctl` `auto-lock-delay` and `clear-session` command are still WIP in trezorctl v0.12.0 and FW v2.3.0, or did I miss read?

```
trezorctl set auto-lock-delay 300
trezorctl clear-session
```

matejcik, 2020-04-17:
@brianddk both have been working fine* on T1 for quite some time. Neither is implemented in TT as of now, i wouldn't even call it "WIP", more like "there is an open issue".

*) for some values of "working fine", mostly depending on what exactly you want ClearSession to do

matejcik, 2020-06-05:
For QA:

Auto-lock is enabled by default and set to 10 minutes. It cannot be completely disabled, but can be configured for up to ~6 days (technical limitation; setting higher values will silently set the maximum).

Minimum is 10 seconds. Specifying more will set (and show) 10 seconds.

Use `trezorctl set auto-lock-delay <seconds>` to set the delay.
Use `trezorctl clear-session` (command name will be changed but for now this works) to lock the device manually.
No need to update trezorctl, these commands existed for T1 for quite some time.

Scenarios:

* Device locks after the specified period of inactivity. A lock screen should be displayed.
* Device can be unlocked by tapping the lock screen.
* Device can be unlocked by initiating an action from wallet software.
* Device can be unlocked by a FIDO/U2F request.
* Work with the device for longer than the specified autolock delay. Ensure that the device doesn't lock while you are using it.
* Let the autolock interrupt an action (leave an address screen on; or signtx confirmation; or just the PIN entry for unlocking). The device should lock, and the wallet software should react with "action was cancelled"
* Let the autolock interrupt FIDO/U2F request. Device should lock, FIDO should be rejected.

This is currently blocked in Suite by https://github.com/trezor/trezor-suite/issues/1971, and there is most likely something similar for old Wallet.

sorooris, 2020-06-08:
@matejcik could you link QA with a non-debug firmware build please? thanks

tsusanka, 2020-06-08:
@sorooris why not get it from CI? From `master` artifact `core fw regular`.

sorooris, 2020-06-08:
@tsusanka thanks, i wasn't sure.

bosomt, 2020-06-19:
currently testing suite branch suite-web/feat/232-pinentry
 https://github.com/trezor/trezor-suite/commit/06d90d21312e090fb2e1b14f83b99be48bd13fb2 with latest 2.3.2 master e630279

1) i often **see enter passphrase dialogue before i see correct dialogue after i unlock** auto-locked device with some confirmation like change pin or name, after less than second it will disappear and I will see correct dialogue, is this firmware or suite issue ?

After quick check I noticed that this happens with 2.3.1 production c6b2580cd FW too and beta version of suite https://github.com/trezor/trezor-suite/commit/06d90d21312e090fb2e1b14f83b99be48bd13fb2

do you recommend to send it as Firmware issue or suite issue ?

2) when trying to **wipe auto-locked device i'm not asked to unlock it** but to confirm wipe, is this intentional ?

tsusanka, 2020-06-19:
> i often see enter passphrase dialogue before i see correct dialogue after i unlock auto-locked device with some confirmation like change pin or name, after less than second it will disappear and I will see correct dialogue, is this firmware or suite issue ?

So you have an auto-locked device and you start some action in Suite (such as change pin). You unlock the device and instead of the change pin dialog you see a passphrase dialog on the device for a brief second? Is that correct? Maybe video could help?

bosomt, 2020-06-19:
@tsusanka I just sent suite issue, screenshot from video and video included
https://github.com/trezor/trezor-suite/issues/2046

strange is  the it only happens once, maybe then passphrase is already cached, i don't know ;)

matejcik, 2020-06-22:
> 1. when trying to **wipe auto-locked device i'm not asked to unlock it** but to confirm wipe, is this intentional ?

yes, this is intentional

tsusanka, 2020-06-29:
So since @bosomt's (1) is not related to the auto-lock (happens on 2.3.1 as well) and (2) is intentional I am closing again. @bosomt were those the only issues you have encountered? In another words, did you successfully test all the other scenarios provided by @matejcik? On `suite-web/feat/232-pinentry`, we still need to test Wallet later

closed 2020-06-29 by tsusanka

bosomt, 2020-06-29:
@tsusanka tested and all features mentioned by matejcik are working fine.

tsusanka, 2020-06-29:
And let's wait for the old Wallet implementation :). I'll let you know when that's ready. Thanks

tsusanka, 2020-06-30:
@trezor/qa please do a quick test with Password Manager as well.

bosomt, 2020-06-30:
@tsusanka i already sent issue 2 days ago for TT 2.3.2 ;)
https://github.com/trezor/trezor-password-manager/issues/95

i will heck it with T1 tomorrow ;)

sorooris, 2020-07-09:
Tested with 2.3.2 14a8da9
[Testpad report](https://satoshilabs.ontestpad.com/script/123/report/?auth=4fab6a3d5572de55fdb5dc3c7421457b)

sorooris, 2020-07-17:
Linked [issue](https://github.com/trezor/trezor-password-manager/issues/95) from the report retested and is fixed by [Connect](https://github.com/trezor/connect/pull/619), removing the _needs QA_ label.

commit e680db9e (2020-05-19): core: implement auto-lock after a configurable timeout (fixes #75)
commit fa8bf306 (2020-05-19): core: implement auto-lock after a configurable timeout (fixes #75)
commit 2f0c74dd (2020-06-04): core: implement auto-lock after a configurable timeout (fixes #75)
PR #1027 (2020-06-04): auto-lock for trezor-core
fixes #75 

this has WAY more commits than it rightly should.

The core functionality is f1c3172, which introduces a global idle timer. It is then possible to register an idle callback, which, after a configurable timeout (we already have autolock timeout fields in storage), engages the lock.

Another big part is a new `loop.spawn` syscall (docs included), which serves as a "controllable" subtask on the application level. All workflow tasks are now `spawn`ed, so that it is possible to shut them down when the autolock engages. (this also answers [this comment](https:// [... cut]