trezor/trezor-firmware #76: add the option to cancel Wipe and SignTx button requests on device display

michal-novacek, 2019-04-17:
After user initiates a wipe or signtx action, they may change their mind and decide to cancel it. However, a cancel button is not displayed anywhere on the device and unplugging the device to cancel the request is inconvenient.

Once a button request is initiated, it is a convention for the web wallet to display a blocking modal that doesn't allow user to navigate elsewhere and doesn't close until the action is finished or cancelled on device. There are ways to close this modal and cancel the action from the web app but they are hacky and go against UX consistency.

tsusanka, 2020-02-28:
I have discussed this with @matejzak and Petr. Petr came up with this for now, which is basically what @prusnak drafted up:

[image]

We have also realized that in some places (such as wipe) a red cancel is probably not a good idea next to a red Hold to confirm button - maybe orange back button or similar?. Also in some cases we might want to leave the inability to cancel the dialog (such as the one when doing backup, see above).

Partially implemented in `tsusanka/hold-to-confirm-cancel`.

tsusanka, 2020-03-02:
Petr has finished and we have agreed on:

[image]

PR is coming.

closed 2020-03-02 by tsusanka

matejcik, 2020-03-02:
wait, shouldn't the "Backup your seed" -> "I understand" be non-cancellable?

right now when you cancel that, you'll probably get a failed backup, isn't that right?

tsusanka, 2020-03-02:
I have found out the "I understand" button is a regular button, not hold to confirm and therefore not affected by this change.

https://github.com/trezor/trezor-firmware/blob/09c3fd1981ced7013c6e5b55530874b4dc5c9e47/core/src/apps/management/reset_device/layout.py#L259

PR #885 (2020-03-02): Add cancel to hold to confirm dialogs
The names ButtonAbort/ButtonCancel are not very good, any suggestions? I didn't come up with anything better than simple ButtonRedCancel and ButtonGreyCancel, which on a second though seems better.

Closes #76. UI can be viewed here: https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/455699058/artifacts/tests/ui_tests/reports/index.html