trezor/trezor-firmware #78: Strange font spacing

good-lly, 2019-04-17:
There is some strange spacing behind letters, for example small 'r' and large 'T'

[image]
[image]

ZdenekSL, 2019-06-25:
@pankrats Please retest this issues.

pankrats, 2019-06-25:
Can confirm, the spacing is still the same. 

Here is the strange spacing after "T":
[image]

And here is the "re" combination (in "remaining)":
[image]

jpochyla, 2019-06-25:
If we decide to fix this, I'd also suggest making the font smaller.

ZdenekSL, 2019-10-24:
Closed after discussion

closed 2019-10-24 by ZdenekSL