trezor/trezor-firmware #88: bn_subtractmod & BN_mod_sub (OpenSSL)

armatusmiles, 2019-04-18:
`
k = 71990190367637002450973982263548715353662720633153471440348968216511857089338
`
`
s =
59304699529132960534443899099941505197933629896569762720193892446897526956229
`
`
curve->order =
115792089237316195423570985008687907852837564279074904382605163141518161494337
`

```
bn_subtractmod(k, &s, &result, &curve->order); ( res = a - b mod prime ) 
```

`
result = 244269669313136432763672053180983025861404219294733517485365402052650653121783
`

```
BN_mod_sub(result, k, s, order, context); ( BN_mod_sub sets r = a - b mod m ) 
```
`
result = 12685490838504041916530083163607210155729090736583708720155075769614330133109
`

The same function, same input parameters different result. Why? 
[BN_mod_sub](https://www.openssl.org/docs/man1.0.2/man3/BN_mod_sub.html)

armatusmiles, 2019-04-18:
I need repeat behavior BN_mod_sub for bignums from trezor

armatusmiles, 2019-04-18:
For some value k for me work this construction:
```
bn_subtractmod(k, &s, &result, &curve->order); 
bn_subtract(&result, &curve->order, &result->s);
```
equals:

```
BN_mod_sub(result, k, s, order, context); ( BN_mod_sub sets r = a - b mod m ) 
```

But for anothers K i need got half_order for get same result:
```
bn_subtractmod(k, &s, &result, &curve->order_half); 
bn_subtract(&result, &curve->order, &result->s);
```

I don't know which check up need? Maybe some one can help me? :)

AmritKumar, 2019-04-19:
Try calling `bn_mod` after `bn_subtractmod`.

closed 2019-04-19 by armatusmiles