trezor/trezor-firmware #90: Trezor freezes in passphrase dialogue

bosomt, 2019-04-20:
After some time you are asked to enter passphrase again.

[image]

When you  select host: dialogue disapears 
Select device: and then hit enter on display: device freezes every time /i was able to repproduce it 4x today/

Latest firmware. Chrome, Nixos

bosomt, 2019-04-22:
and device freezes even when you enter passphrase and then hit enter, so it's not problem with empty passphrase

bosomt, 2019-04-22:
same thing is happening in host/device screen selection.
Just hold your screen touched and device will freeze after some time

bosomt, 2019-04-24:
> same thing is happening in host/device screen selection.
> Just hold your screen touched and device will freeze after some time.
> 
> And this has nothing to do with some sleep time, this happens after device is connected

same in 2.1.1 FW version
connect, hold finger on touchscreen
device freezes in 5 sec

bosomt, 2019-04-25:
and when you wait long enough then when you select host/device then display goes off and device freezes

tsusanka, 2019-04-26:
@bosomt please find out how this works with emulator

bosomt, 2019-04-26:
im on it

bosomt, 2019-04-28:
i just found out that device doesn't really freezes.
If you send different request to device it will wake up and shows new dialogue and it works fine after that
/for example if it freezes in passphrase dialogue and you send show full address request/

matejcik, 2019-05-09:
AFAICT this is two distinct issues

issue 1: wallet sometimes doesn't send `PassphraseStateAck`.
to reproduce:
1. open wallet
2. enter passphrase "a" on device
3. switch currency
4. enter any passphrase on device
5. switch back to first currency
6. enter passphrase "d" on device

This triggers on wallet.trezor.io but not on beta wallet. Presumably this is only one way to trigger a particular state, another is @bosomt's thing with waiting for passphrase prompt, which I didn't manage to reproduce.

Can be reproduced on emulator.

This will freeze after pressing Enter on the passphrase keyboard. Also this will be cleared when you start another action on the device.

Should be handled separately by wallet team.

issue 2: when you hold a button, sometimes trezor stops responding. I didn't manage to reproduce on emulator. I'm closing this issue in favor of #144.

closed 2019-05-09 by matejcik