trezor/trezor-firmware #94: Firmware error reported when signing Cardano transactions with many inputs

tokyotechlab, 2019-04-20:
When I tried to sweep ADA using Trezor, I received a firmware error.
The problem seems to arise when the number of inputs I'm trying to spend exceed 7 or 8. Per matejcik's suggestion, I'm creating an issue in trezor-firmware.

I was able to reproduce the error on both adalite and trezorctl.
Trezorctl version: cloned from github on 4/15
Trezor firmware version: 2.1.0

Please see the repro files and the error message below.

tokyotechlab, 2019-04-20:
This fails.

{"inputs": [{"address_n": [2147483692, 2147485463, 2147483649, 0, 2], "path": "m/44'/1815'/1'/0/2", "prev_hash": "77d18f5aec0e1e1d06618fb5063b5304d7a366292afc560809691889f556e088", "prev_index": 5, "type": 0}, {"address_n": [2147483692, 2147485463, 2147483649, 0, 2], "path": "m/44'/1815'/1'/0/2", "prev_hash": "77d18f5aec0e1e1d06618fb5063b5304d7a366292afc560809691889f556e088", "prev_index": 4, "type": 0}, {"address_n": [2147483692, 2147485463, 2147483649, 0, 2], "path": "m/44'/1815'/1'/0/2", "prev_hash": "77d18f5aec0e1e1d06618fb5063b5304d7a366292afc560809691889f556e088", "prev_index": 7, "type": 0}, {"address_n": [2147483692, 2147485463, 2147483649, 0, 2], "path": "m/44'/1815'/1'/0/2", "prev_hash": "77d18f5aec0e1e1d06618fb5063b5304d7a366292afc560809691889f556e088", "prev_index": 6, "type": 0}, {"address_n": [2147483692, 2147485463, 2147483649, 0, 2], "path": "m/44'/1815'/1'/0/2", "prev_hash": "77d18f5aec0e1e1d06618fb5063b5304d7a366292afc560809691889f556e088", "prev_index": 17, "type": 0}, {"address_n": [2147483692, 2147485463, 2147483649, 0, 2], "path": "m/44'/1815'/1'/0/2", "prev_hash": "77d18f5aec0e1e1d06618fb5063b5304d7a366292afc560809691889f556e088", "prev_index": 1, "type": 0}, {"address_n": [2147483692, 2147485463, 2147483649, 0, 2], "path": "m/44'/1815'/1'/0/2", "prev_hash": "77d18f5aec0e1e1d06618fb5063b5304d7a366292afc560809691889f556e088", "prev_index": 16, "type": 0}, {"address_n": [2147483692, 2147485463, 2147483649, 0, 2], "path": "m/44'/1815'/1'/0/2", "prev_hash": "77d18f5aec0e1e1d06618fb5063b5304d7a366292afc560809691889f556e088", "prev_index": 0, "type": 0}, {"address_n": [2147483692, 2147485463, 2147483649, 0, 2], "path": "m/44'/1815'/1'/0/2", "prev_hash": "77d18f5aec0e1e1d06618fb5063b5304d7a366292afc560809691889f556e088", "prev_index": 19, "type": 0}, {"address_n": [2147483692, 2147485463, 2147483649, 0, 2], "path": "m/44'/1815'/1'/0/2", "prev_hash": "77d18f5aec0e1e1d06618fb5063b5304d7a366292afc560809691889f556e088", "prev_index": 3, "type": 0}, {"address_n": [2147483692, 2147485463, 2147483649, 0, 2], "path": "m/44'/1815'/1'/0/2", "prev_hash": "77d18f5aec0e1e1d06618fb5063b5304d7a366292afc560809691889f556e088", "prev_index": 18, "type": 0}, {"address_n": [2147483692, 2147485463, 2147483649, 0, 2], "path": "m/44'/1815'/1'/0/2", "prev_hash": "77d18f5aec0e1e1d06618fb5063b5304d7a366292afc560809691889f556e088", "prev_index": 2, "type": 0}], "outputs": [{"address": "Ae2tdPwUPEZKDfYqhRvAp4iyJZhmosy38yt5eztQxwReFPm1PzgJoQxXL6F", "amount": 1000000}, {"address": "Ae2tdPwUPEZLhFtRtWGc1hGxNaigXmadDyNQN2Tt1UMoNnyF5ESb7JHsoEC", "amount": 100000}], "transactions": ["839f8200d81858248258202598eba19c68ea3be11c670fcc01692d2139adaf5656d923621b6559de38da5c098200d81858248258202598eba19c68ea3be11c670fcc01692d2139adaf5656d923621b6559de38da5c0a8200d81858248258202598eba19c68ea3be11c670fcc01692d2139adaf5656d923621b6559de38da5c0b8200d818582482582026a42df8d26218090b456317f7a5a864ac04a0e9d395112747ffc26e71e0852e00ff9f8282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581c0486c9fe549de2179467ee360c041d015a06c3df5c0da43b49133fd4a0001ad97aaa771a00231860ffa0", "839f8200d81858248258202598eba19c68ea3be11c670fcc01692d2139adaf5656d923621b6559de38da5c098200d81858248258202598eba19c68ea3be11c670fcc01692d2139adaf5656d923621b6559de38da5c0a8200d81858248258202598eba19c68ea3be11c670fcc01692d2139adaf5656d923621b6559de38da5c0b8200d818582482582026a42df8d26218090b456317f7a5a864ac04a0e9d395112747ffc26e71e0852e00ff9f8282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67 [... cut ...] 6a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581c0486c9fe549de2179467ee360c041d015a06c3df5c0da43b49133fd4a0001ad97aaa771a00231860ffa0"]}

tokyotechlab, 2019-04-20:
This works fine.

{"inputs": [{"address_n": [2147483692, 2147485463, 2147483649, 0, 2], "path": "m/44'/1815'/1'/0/2", "prev_hash": "77d18f5aec0e1e1d06618fb5063b5304d7a366292afc560809691889f556e088", "prev_index": 11, "type": 0}, {"address_n": [2147483692, 2147485463, 2147483649, 0, 2], "path": "m/44'/1815'/1'/0/2", "prev_hash": "77d18f5aec0e1e1d06618fb5063b5304d7a366292afc560809691889f556e088", "prev_index": 10, "type": 0}, {"address_n": [2147483692, 2147485463, 2147483649, 0, 2], "path": "m/44'/1815'/1'/0/2", "prev_hash": "2598eba19c68ea3be11c670fcc01692d2139adaf5656d923621b6559de38da5c", "prev_index": 12, "type": 0}], "outputs": [{"address": "Ae2tdPwUPEZKDfYqhRvAp4iyJZhmosy38yt5eztQxwReFPm1PzgJoQxXL6F", "amount": 100000}, {"address": "Ae2tdPwUPEZLhFtRtWGc1hGxNaigXmadDyNQN2Tt1UMoNnyF5ESb7JHsoEC", "amount": 100000}], "transactions": ["839f8200d81858248258202598eba19c68ea3be11c670fcc01692d2139adaf5656d923621b6559de38da5c098200d81858248258202598eba19c68ea3be11c670fcc01692d2139adaf5656d923621b6559de38da5c0a8200d81858248258202598eba19c68ea3be11c670fcc01692d2139adaf5656d923621b6559de38da5c0b8200d818582482582026a42df8d26218090b456317f7a5a864ac04a0e9d395112747ffc26e71e0852e00ff9f8282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581c0486c9fe549de2179467ee360c041d015a06c3df5c0da43b49133fd4a0001ad97aaa771a00231860ffa0", "839f8200d81858248258202598eba19c68ea3be11c670fcc01692d2139adaf5656d923621b6559de38da5c098200d81858248258202598eba19c68ea3be11c670fcc01692d2139adaf5656d923621b6559de38da5c0a8200d81858248258202598eba19c68ea3be11c670fcc01692d2139adaf5656d923621b6559de38da5c0b8200d818582482582026a42df8d26218090b456317f7a5a864ac04a0e9d395112747ffc26e71e0852e00ff9f8282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054c [... cut ...] 6a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581cef67b796b029c75054ccab5a2d7fb0086715e97b56dec84b5679d569a0001a2163b1651a000186a08282d818582183581c0486c9fe549de2179467ee360c041d015a06c3df5c0da43b49133fd4a0001ad97aaa771a0171acf4ffa0"]}

tokyotechlab, 2019-04-20:
[2019-04-20 23:47:51,358] trezorlib.transport.protocol DEBUG: sending message: CardanoSignTx
CardanoSignTx (886 bytes) {
    inputs: [
        CardanoTxInputType (60 bytes) {
            address_n: [2147483692, 2147485463, 2147483649, 0, 2],
            prev_hash: 32 bytes 0x77d18f5aec0e1e1d06618fb5063b5304d7a366292afc560809691889f556e088,
            prev_index: 5,
            type: 0,
        },
        CardanoTxInputType (60 bytes) {
            address_n: [2147483692, 2147485463, 2147483649, 0, 2],
            prev_hash: 32 bytes 0x77d18f5aec0e1e1d06618fb5063b5304d7a366292afc560809691889f556e088,
            prev_index: 4,
            type: 0,
        },
        CardanoTxInputType (60 bytes) {
            address_n: [2147483692, 2147485463, 2147483649, 0, 2],
            prev_hash: 32 bytes 0x77d18f5aec0e1e1d06618fb5063b5304d7a366292afc560809691889f556e088,
            prev_index: 7,
            type: 0,
        },
        CardanoTxInputType (60 bytes) {
            address_n: [2147483692, 2147485463, 2147483649, 0, 2],
            prev_hash: 32 bytes 0x77d18f5aec0e1e1d06618fb5063b5304d7a366292afc560809691889f556e088,
            prev_index: 6,
            type: 0,
        },
        CardanoTxInputType (60 bytes) {
            address_n: [2147483692, 2147485463, 2147483649, 0, 2],
            prev_hash: 32 bytes 0x77d18f5aec0e1e1d06618fb5063b5304d7a366292afc560809691889f556e088,
            prev_index: 17,
            type: 0,
        },
        CardanoTxInputType (60 bytes) {
            address_n: [2147483692, 2147485463, 2147483649, 0, 2],
            prev_hash: 32 bytes 0x77d18f5aec0e1e1d06618fb5063b5304d7a366292afc560809691889f556e088,
            prev_index: 1,
            type: 0,
        },
        CardanoTxInputType (60 bytes) {
            address_n: [2147483692, 2147485463, 2147483649, 0, 2],
            prev_hash: 32 bytes 0x77d18f5aec0e1e1d06618fb5063b5304d7a366292afc560809691889f556e088,
            prev_index: 16,
            type: 0,
        },
        CardanoTxInputType (60 bytes) {
            address_n: [2147483692, 2147485463, 2147483649, 0, 2],
            prev_hash: 32 bytes 0x77d18f5aec0e1e1d06618fb5063b5304d7a366292afc560809691889f556e088,
            prev_index: 0,
            type: 0,
        },
        CardanoTxInputType (60 bytes) {
            address_n: [2147483692, 2147485463, 2147483649, 0, 2],
            prev_hash: 32 bytes 0x77d18f5aec0e1e1d06618fb5063b5304d7a366292afc560809691889f556e088,
            prev_index: 19,
            type: 0,
        },
        CardanoTxInputType (60 bytes) {
            address_n: [2147483692, 2147485463, 2147483649, 0, 2],
            prev_hash: 32 bytes 0x77d18f5aec0e1e1d06618fb5063b5304d7a366292afc560809691889f556e088,
            prev_index: 3,
            type: 0,
        },
        CardanoTxInputType (60 bytes) {
            address_n: [2147483692, 2147485463, 2147483649, 0, 2],
            prev_hash: 32 bytes 0x77d18f5aec0e1e1d06618fb5063b5304d7a366292afc560809691889f556e088,
            prev_index: 18,
            type: 0,
        },
        CardanoTxInputType (60 bytes) {
[... 138 lines cut ...]
  File "/usr/local/bin/trezorctl", line 1989, in <module>
    cli()  # pylint: disable=E1120
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/click/decorators.py", line 27, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "/usr/local/bin/trezorctl", line 1441, in cardano_sign_tx
    signed_transaction = cardano.sign_tx(client, inputs, outputs, transactions, network)
  File "/usr/local/lib/python3.6/dist-packages/trezorlib/tools.py", line 233, in wrapped_f
    return f(client, *args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/trezorlib/cardano.py", line 60, in sign_tx
    response = client.call(ack_message)
  File "/usr/local/lib/python3.6/dist-packages/trezorlib/tools.py", line 233, in wrapped_f
    return f(client, *args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/trezorlib/client.py", line 196, in call
    raise exceptions.TrezorFailure(resp)
trezorlib.exceptions.TrezorFailure: FirmwareError: Firmware error

tsusanka, 2019-04-21:
@refi93, @ddeath do you think you could take a look at this?

refi93, 2019-04-21:
@tokyotechlab @tsusanka we are working on a fix that should increase the maximum number of inputs in the transaction. We will make a PR soon, probably this week. We will try to improve it further.

The problem is that currently the transaction to be signed is held in-memory while it's being signed/constructed together with the previous transactions, which serve to verify the amounts of ADA in the inputs, and that limits the size of the transaction given Trezor's restricted memory. 7-8 inputs per transaction are very little though and it is the result of not managing the memory properly - the limit should be definitely raisable. We are already working on optimizing the memory consumption of the Cardano integration in Trezor. 

A proper fix would be processing the transaction in chunks while it's being sent to Trezor, but this would involve a probably significant refactor which is questionable if it is worth it, since the upcoming Cardano "Shelley" update should involve a significant change of the format of serialization of the transactions as well as eliminate the need to check previous transactions (by adding self describing transaction inputs), in general, it should make things a lot easier. The exact date was not disclosed by Cardano devs, unfortunately, but should be in following months.

refi93, 2019-04-26:
sorry for the delay, the PR should be ready the next week

closed 2019-05-07 by matejcik

octox-io, 2019-07-16:
Hi, I still have this issue with my Trezor-T and Adalite wallet.
Currently, I'm not able to send my ADA out from my wallet, despite upgrading to latest Trezor firmware.
Any help ?

ZdenekSL, 2019-07-17:
@octox-io Hello there, what FW do you use? We fixed this in 2.1.1, so if you have older firmware, please try to update your device. If this doesn't help, I would like to ask you to contact our support: https://trezor.io/support/. They are highly skilled and trained people, who will help you to solve your problem. If they will not find any solution, they will escalate the issue with our firmware team.

Thank you

pankrats, 2019-08-06:
I checked the issue with the upcoming firmware 2.1.3 and looks like it is fixed. No problems there when signing a transaction with 7 or 8 inputs.

commit afdfd5fb (2019-05-07): cardano: optimize transaction signing

squash merge. closes #94 and #130

Squashed commit of the following:

commit 1adc68051df863f5ce3767f6594d91fc9ad7bb16
Author: refi93 <rafael.korbas@gmail.com>
Date:   Tue May 7 00:13:07 2019 +0200

    cr fix

commit 833b831754aa3dfce4630e9d64e7fb6c5147ceb1
Author: refi93 <rafael.korbas@gmail.com>
Date:   Mon May 6 21:59:49 2019 +0200

    cr fix style

commit 2962e697c993d3df9a53dcc6cebc9148c1d701f5
Author: Matúš Juran <matus.juran@vacuumlabs.com>
Date:   Thu May 2 16:57:55 2019 +0200

    cardano: optimize transaction signing

    Optimize the storage o [... cut]