trezor/trezor-firmware #96: Add more test vectors for Schnorr

prusnak, 2019-04-21:
In https://github.com/trezor/trezor-firmware/commit/8114e0f9fc7af5d41ac73d69209b4065d3638327 we added Schnorr signatures.

Let's add more test vectors to `test_check.c`; for example:

* https://gist.github.com/kallewoof/5d623445802a84f17cc7ff5572109074
* https://raw.githubusercontent.com/sipa/bips/bip-schnorr/bip-schnorr/test-vectors.csv

tsusanka, 2019-04-26:
@onvej-sl would like to do this one?

onvej-sl, 2019-05-02:
I will.

Sajjon, 2019-05-03:
[Pieter Wuille aka "sipa"](https://github.com/sipa) created some test vectors in the [Schnorr BIP](https://github.com/sipa/bips/blob/bip-schnorr/bip-schnorr.mediawiki#test-vectors), you can [find the vectors here](https://github.com/sipa/bips/blob/bip-schnorr/bip-schnorr/test-vectors.csv)

I used them in [my Schnorr implementation in Swift](https://github.com/Sajjon/EllipticCurveKit/blob/equations/Source/EllipticCurve/Signing/CommonSigning/Schnorr.swift)

onvej-sl, 2019-05-06:
Suggested test vectors are not compatible with our implementation. I need to search for different ones. 
<hr>

Assume:
  * G is a group generator
  * H is a hash function
  * M is a message
  * k is a private key, K = k*G is the corresponding public key
 
Let:
  * r be a private nonce, R = r*G be the corresponding "public nonce"
  * h = H(R || M)
  * s = r - k*h
  * h' = H(R || K || M)
  * s' = r - k*h'

There are four basic types of Schnorr signature:
  1. (s, h)
  2. (s, R)
  3. (s', h')
  4. (s', R)

Signatures of the types 1 and 2 can be aggregated, the other two cannot. Signatures of the types 2 and 4 support faster batch verification.

https://github.com/trezor/trezor-firmware/commit/8114e0f9fc7af5d41ac73d69209b4065d3638327 implements a signature of the type 3. Both suggested set of test vectors are of the type 4.

Sajjon, 2019-05-06:
You could just fork my [implementation in EllipticCurveKit](https://github.com/Sajjon/EllipticCurveKit/blob/master/Source/EllipticCurve/Signing/CommonSigning/Schnorr.swift) and use as reference together with an adjusted version of [Sipa's Python reference code](https://github.com/sipa/bips/blob/bip-schnorr/bip-schnorr/reference.py), if the two of them results in the same output at least you have some reference vectors.

onvej-sl, 2019-05-14:
Tests already seem to cover the most obvious edge cases. Postponed until [Schnorr signatures for bitcoin](https://github.com/sipa/bips/blob/bip-schnorr/bip-schnorr.mediawiki) are implemented.

Closing.

closed 2019-05-14 by onvej-sl