trezor/trezor-firmware #98: [QUESTION]Bug at RDP verification for option bits?

denisacostaq, 2019-04-22:
I'm trying to figure out how this code works:
https://github.com/trezor/trezor-firmware/blob/b4cec0ae541adca38ccfc46ffe6ec6ff0a3038fb/legacy/memory.c#L35-L39

According to the above documentation should it be like this?:
```c
uint8_t memory_protection_level(void) {
	flash_unlock();
	flash_unlock_option_bytes();
	uint32_t volatile *opts_register = (uint32_t*)FLASH_OPTCR;
	// masking bits in 15:8 range
	uint32_t mask = 0B00000000000000001111111100000000;
	uint8_t ret = 1;
	switch (((*opts_register) & mask) >> 8) {
		case 0xAA:
			ret = 0;
			break;
		case 0xCC:
			ret = 2;
			break;
		default:
			ret = 1;
			break;
	}
	flash_lock_option_bytes();
	flash_lock();
	return ret;
}
```

Can someone please help me to understand what I'm missing.

prusnak, 2019-04-23:
What is your question exactly?

prusnak, 2019-04-23:
No need to call `flash_unlock(); flash_unlock_option_bytes(); flash_lock_option_bytes(); flash_lock();` if you do only reading.

Otherwise, it seems OK.

closed 2019-04-23 by prusnak