trezor/trezor-firmware #108: CI improvements

tsusanka, 2019-04-25:
- [ ] Propagate tests results into GitHub.
- [ ] Add python tests 
  - @matejcik I wasn't sure if we can/want run tox inside pipenv, so I'm leaving this task on you
- [ ] Cache python's pipenv/virtualenv cache.
- [ ] Cache submodules/vendor.
- [ ] Make sure Pull Requests from external contributors are ran on CI
  - Because I think they are not at the moment.
- [ ] Split .gitlab-ci into several files using `include`.
- [ ] Investigate if `ITERS=10` is run under pipenv (see 'test crypto' job)
- [ ] Can we remove setup.cfg's tool:pytest section?
  - @matejcik due to this setup it was impossible to run tests in crypto/, because `--pyargs trezorlib.tests.device_tests` is always attached. Is it safe to remove it completely?
- [ ] Add "matrix" tests based on travis.
  - In other words, some combinations are not tested currently.
- [ ] Propagate full tests results log into GitHub for external contributors.
- [ ] Add sanitizer tools (tracked in #44)
- [ ] Add debug firmware builds to Gitlab (confirm we actually want this).

tsusanka, 2019-04-28:
> Replace manual "environment" build with automatic one. Docker should know quite quickly it doesn't have to be updated.

We have to think this one through. Although it is true that the job is pretty quick, in the current settings it tags the build as `$CONTAINER_NAME:$CI_COMMIT_SHA`, in other words using the commit hash. That means that it gets uploaded into the registry every time, which could bloat pretty quickly. So we need to tag it differently or come up with another solution.

tsusanka, 2019-04-29:
Note: to force pipenv to cache files in the current folder instead of HOME we need to use both `PIPENV_CACHE_DIR="dir"` and `PIPENV_VENV_IN_PROJECT=true`.

matejcik, 2019-05-07:
i removed the `setup.cfg` in core, and left a `pytest.ini` in its place for convenient running of device tests from the `core` subdir. The global setup.cfg has configs for pyflake and isort.

python-trezor still has its own setup.cfg because there are slight differences and I don't think it's possible to inherit the identical parts from parent directory.

closed 2019-09-18 by tsusanka