trezor/trezor-firmware #112: Flash firmware into trezor model t,encountered "Error: open failed "

iroan, 2019-04-29:
I want to flash firmware i made into trezor model t,but when i exec command,error encountered:
```shell
    (trezor-core) ➜  trezor-core git:(own) ✗ make flash
    openocd -f interface/stlink-v2.cfg -c "transport select hla_swd" -f target/stm32f4x.cfg -c "init; reset halt; flash write_image erase build/boardloader/boardloader.bin 0x08000000; exit"
    Open On-Chip Debugger 0.10.0
    Licensed under GNU GPL v2
    For bug reports, read
    	http://openocd.org/doc/doxygen/bugs.html
    hla_swd
    Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
    adapter speed: 2000 kHz
    adapter_nsrst_delay: 100
    none separate
    Info : Unable to match requested speed 2000 kHz, using 1800 kHz
    Info : Unable to match requested speed 2000 kHz, using 1800 kHz
    Info : clock speed 1800 kHz
    Error: open failed
    in procedure 'init'
    in procedure 'ocd_bouncer'
    
    make: *** [flash_boardloader] Error 1
    (trezor-core) ➜  trezor-core git:(own) ✗
```
env:
- macOS Mojave version 10.14.4
- [trezor-core](https://github.com/trezor/trezor-core.git),base on the v2.1.0
- openocd  v0.10.0

google "Error: open failed in procedure 'init' in procedure 'ocd_bouncer'",i find [a post](https://forum.bitcraze.io/viewtopic.php?t=2679)but I only have one USB B to USB C cable. 
- Do I have to have a st-link?
- How i can flash custom firmware?

Any help would be really appreciated!

prusnak, 2019-04-29:
You just need to use `trezorctl firmware-update -f your_firmware.bin` like you would with any other firmware update.

`make flash` uses `openocd` which uses ST-LINK to upload the firmware to pre-production devices.

closed 2019-04-29 by prusnak

iroan, 2019-04-29:
It worked!  Thank you very very much！ @prusnak