trezor/trezor-firmware #113: Passphrase dialogue after disabling PIN

sorooris, 2019-04-29:
TT (2.1.1), passphrase and PIN enabled
Chromium, any wallet

After successfully setting up a PIN on a device and leaving the onboarding process, disabling the PIN in the advanced settings tab, a page dialogue and a device prompt for passphrase entry are displayed. Could be reproduced only sometimes (after clearing cache/storage) and refreshing the page.
1) Selecting Host entry hides the dialogue and continues normally.
2) Selecting Device entry and entering a correct passphrase continues normally.
3) Selecting Device entry and entering an incorrect passphrase freezes the device until another prompt is sent to the device.
[image]

prusnak, 2019-04-29:
This does not seem like a firmware issue, but rather a web wallet issue.

tsusanka, 2019-04-29:
Might be connected to https://github.com/trezor/trezor-firmware/issues/75#issuecomment-486690410

sorooris, 2019-04-30:
Closing and moving into the wallet repo (https://github.com/satoshilabs/mytrezor/issues/753).

closed 2019-04-30 by sorooris