trezor/trezor-firmware #119: Add Cosmos ( ATOM ) mainnet support

megumin2019, 2019-04-29:
also add support for binance chain.

matejcik, 2019-04-30:
This is not a forum for coin support requests. If you wish, you can contribute code yourself, please follow instructions at https://wiki.trezor.io/Developers_guide:Contributing

binance chain is in active development

xloem, 2023-08-20:
@matejcik Has this situation changed four years later?

Where can the binance work be found?

Is there a checklist anywhere of what coins are still to be ported into Trezor Suite to open capacity for new coin families?

Cosmos is a large group of small delegated proof of stake chains, each generally mined by a small group of validators with the largest stakes. Transaction signing is cryptographically similar to bitcoin chains but the payload is google protobufs formatted.

Cosmos users are using the Ledger and Keystone hardware wallets, but there has been complaint regarding Ledger and people want to try the Trezor.

I'm noting it may be appropriate to add coins as forks depending on Trezor's situation.

closed 2023-08-27 by Hannsek