trezor/trezor-firmware #126: Add firmware update scenario to CI

prusnak, 2019-05-01:
To catch issues like https://github.com/trezor/trezor-firmware/issues/122 which do not appear unless the storage has the format of a previous firmware release.

Once https://github.com/trezor/trezor-firmware/issues/125 is implemented we can cache emulator binaries of older firmware builds and perform tests across two versions.

Another option is to add older storage format in the unit tests and try importing that into a currently built firmware release.

prusnak, 2019-05-24:
Most of this is already done.

tsusanka, 2019-05-29:
Patching the library seems not enough to run the binaries on [CI](https://gitlab.corp.sldev.cz/trezor/trezor-firmware/-/jobs/25711). CI runs on Debian Stretch and there is some glibc mismatch:

`/lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.25' not found`

I believe the best way to solve this is to rebuild the binaries using docker (with the same image as is on CI or do it directly there).

cc @1000101

tsusanka, 2019-06-07:
Discussed with admins that:

- firmware-dev website will be locked for internal use
- We'll copy the current emulator binaries from trezor-public/artificats to this page
- I'll finish the CI job, which tests the upgrades. It will not use the trezor-public/artificats repository as a submodule, rather it wil download these binaries from this page.
- Also, I'll add a job, which in case a commit is tagged will build the emulator and  upload to this website for further use.

1000101, 2019-06-07:
> Discussed with admins that:
> 
>     * firmware-dev website will be locked for internal use
> 
TBD
>     * We'll copy the current emulator binaries from trezor-public/artificats to that page
> 
Done.

tsusanka, 2019-06-13:
I've rebuild the core builds again, because of https://github.com/trezor/trezor-firmware/commit/779ff86214eb6856ee0cb1e6aca697b006ab9e11 . Otherwise, this is almost done apart from:
- [ ] solve matejcik's point on version check (https://github.com/trezor/trezor-firmware/pull/214#discussion_r289803698)
- [ ] solve how to pass authentication details or how to download from the server directly (TBD with admins)
- [ ] put on CI
- [ ] profit

tsusanka, 2019-07-30:
Unless, there are some specific reasons this assigned to @matejcik, I believe I should finish this one.

matejcik, 2019-07-30:
i don't know of anything, please go ahead :)

commit 077735f5 (2019-08-26): ci: introduce upgrade tests

closes #126
PR #430 (2019-08-26): ci: introduce upgrade tests (WIP)
This introduces upgrade tests. The Emulator Wrapper and python script were written by @prusnak originally with some modifications of mine. The emulators are built on CI and stored on our [internal dev server](https://firmware.corp.sldev.cz/upgrade_tests/).

All versions that are found on the website are tested. I will also add a CI job, which will deploy new emulators to this website if a new tag occurs (meaning anytime we tag a new version, it will be deployed there automatically and therefore upgrade-tested). This closes #126.

----

We need to:

- [x] Revie [... cut]
closed 2019-08-26 by tsusanka