trezor/trezor-firmware #132: Detect "variable referenced before assignment"

prusnak, 2019-05-03:
This should be easy even without using static analyzer such as MyPy.

prusnak, 2019-05-03:
`NameError: local variable referenced before assignment`

We could avoid bugs like https://github.com/trezor/trezor-firmware/issues/122

prusnak, 2019-05-03:
Just figured out this is not easy at all as the problem is much more complicated. Let's try to finish typing ASAP.

matejcik, 2019-05-04:
let's keep this open for the purpose of enabling at least some pylint warnings

prusnak, 2019-05-04:
OK. Just a note: error in #122 would not be caught by PyLint, it could've been discovered only via typing.

tsusanka, 2019-09-06:
The current upgrade bug from 2.0.10 to 2.1.4/master where we access a const() variable with an underscore in its name (see [here](https://github.com/trezor/trezor-firmware/blob/core/v2.1.4/core/src/apps/common/storage/__init__.py#L30
)), might have been caught with pylint.

tsusanka, 2020-11-23:
It would be nice to do this iteratively. That means put pytlint into CI with a few checks and add them gradually.

hynek-jina, 2021-12-17:
@matejcik

matejcik, 2021-12-17:
pylint was enabled in #1848

closed 2021-12-17 by matejcik