trezor/trezor-firmware #133: Trezor t unable to sign legacy multisig inputs

achow101, 2019-05-04:
In my testing of the Trezor T with HWI, I found that I was unable to sign transactions containing legacy multisig inputs. It seemed that what was happening was that the same input was being signed twice and another input was not being signed.

The same code works for Trezor 1 and is able to sign a transaction using the Trezor 1.

matejcik, 2019-05-04:
can you provide a sample transaction input?

achow101, 2019-05-04:
It looks like the issue is actually with mixed segwit and non-segwit inputs. The multisig transaction I was signing earlier had both a non-segwit multisig and a segwit one. Removing the non-segwit input made it work. The non-segwit input by itself could be signed.

To make debugging easier, I made a different transaction with mixed segwit and non-segwit that is not multisig. Here is the PSBT for that. (This is on regtest):
```
cHNidP8BAHsCAAAAApTd2qYtyLLk2aVAW0pj6Z3SD11hroeNzkL0k88hDDQdAQAAAAD/////lN3api3IsuTZpUBbSmPpndIPXWGuh43OQvSTzyEMNB0CAAAAAP////8B7K3rCwAAAAAWABQCrNTzocb2NYP9Ohj3SPStuS319AAAAAAAAQCTAgAAAAHNfKrAGwgkIJi77nOCJIkduTBTr9vO+7tCjdsOihcc3gEAAAAA/v///wMo+cW4AAAAABYAFHsF7pIVY3SsJRyZ+1kG9XX7lgLJAOH1BQAAAAAZdqkUnjlApR3S/wdTSYIp+SHSeJ7zUzCIrADh9QUAAAAAFgAU/FNL+qz2/YHN7AkHaXFmY+uuXlgAAAAAIgYCwZomYuU+DmBsT2Cl9vLuS2Osv4rL3UvI6G/U+rprl7AYBciwBywAAIABAACAAAAAgAAAAAAJAAAAAAEBHwDh9QUAAAAAFgAU/FNL+qz2/YHN7AkHaXFmY+uuXlgiBgNIdJ5HVq7cLrcpFMtqrh9nY5TRbdThJL7dTDVuEz+f5xgFyLAHLAAAgAEAAIAAAACAAAAAAAgAAAAAAA==

{
  "tx": {
    "txid": "e7b45f4ac5eeadd0f2e0d97702ccc486d593254f3615dcf72a5d3ce3be00be19",
    "hash": "e7b45f4ac5eeadd0f2e0d97702ccc486d593254f3615dcf72a5d3ce3be00be19",
    "version": 2,
    "size": 123,
    "vsize": 123,
    "weight": 492,
    "locktime": 0,
    "vin": [
      {
        "txid": "1d340c21cf93f442ce8d87ae615d0fd29de9634a5b40a5d9e4b2c82da6dadd94",
        "vout": 1,
        "scriptSig": {
          "asm": "",
          "hex": ""
        },
        "sequence": 4294967295
      },
      {
        "txid": "1d340c21cf93f442ce8d87ae615d0fd29de9634a5b40a5d9e4b2c82da6dadd94",
        "vout": 2,
        "scriptSig": {
          "asm": "",
          "hex": ""
        },
        "sequence": 4294967295
      }
    ],
    "vout": [
      {
        "value": 1.99994860,
        "n": 0,
        "scriptPubKey": {
          "asm": "0 02acd4f3a1c6f63583fd3a18f748f4adb92df5f4",
          "hex": "001402acd4f3a1c6f63583fd3a18f748f4adb92df5f4",
          "reqSigs": 1,
          "type": "witness_v0_keyhash",
          "addresses": [
            "bcrt1qq2kdfuapcmmrtqla8gv0wj854kujma054d4d33"
          ]
        }
      }
    ]
  },
  "unknown": {
  },
  "inputs": [
    {
      "non_witness_utxo": {
        "txid": "1d340c21cf93f442ce8d87ae615d0fd29de9634a5b40a5d9e4b2c82da6dadd94",
        "hash": "1d340c21cf93f442ce8d87ae615d0fd29de9634a5b40a5d9e4b2c82da6dadd94",
        "version": 2,
        "size": 147,
        "vsize": 147,
        "weight": 588,
        "locktime": 0,
        "vin": [
          {
            "txid": "de1c178a0edb8d42bbfbcedbaf5330b91d89248273eebb982024081bc0aa7ccd",
            "vout": 1,
            "scriptSig": {
              "asm": "",
              "hex": ""
            },
            "sequence": 4294967294
          }
        ],
        "vout": [
          {
            "value": 30.99982120,
            "n": 0,
            "scriptPubKey": {
              "asm": "0 7b05ee92156374ac251c99fb5906f575fb9602c9",
              "hex": "00147b05ee92156374ac251c99fb5906f575fb9602c9",
              "reqSigs": 1,
              "type": "witness_v0_keyhash",
              "addresses": [
                "bcrt1q0vz7ays4vd62cfgun8a4jph4whaevqkfnewm2u"
              ]
            }
          },
          {
            "value": 1.00000000,
            "n": 1,
            "scriptPubKey": {
              "asm": "OP_DUP OP_HASH160 9e3940a51dd2ff0753498229f921d2789ef35330 OP_EQUALVERIFY OP_CHECKSIG",
              "hex": "76a9149e3940a51dd2ff0753498229f921d2789ef3533088ac",
              "reqSigs": 1,
              "type": "pubkeyhash",
              "addresses": [
                "muwZeLZMGbT11GjZfrbmwPHsHURtpXkW53"
              ]
            }
          },
          {
            "value": 1.00000000,
            "n": 2,
            "scriptPubKey": {
              "asm": "0 fc534bfaacf6fd81cdec090769716663ebae5e58",
              "hex": "0014fc534bfaacf6fd81cdec090769716663ebae5e58",
              "reqSigs": 1,
              "type": "witness_v0_keyhash",
              "addresses": [
                "bcrt1ql3f5h74v7m7crn0vpyrkjutxv046uhjc9jducy"
              ]
            }
          }
        ]
      },
      "bip32_derivs": [
        {
          "pubkey": "02c19a2662e53e0e606c4f60a5f6f2ee4b63acbf8acbdd4bc8e86fd4faba6b97b0",
          "master_fingerprint": "05c8b007",
          "path": "m/44'/1'/0'/0/9"
        }
      ]
    },
    {
      "witness_utxo": {
        "amount": 1.00000000,
        "scriptPubKey": {
          "asm": "0 fc534bfaacf6fd81cdec090769716663ebae5e58",
          "hex": "0014fc534bfaacf6fd81cdec090769716663ebae5e58",
          "type": "witness_v0_keyhash",
          "address": "bcrt1ql3f5h74v7m7crn0vpyrkjutxv046uhjc9jducy"
        }
      },
      "bip32_derivs": [
        {
          "pubkey": "0348749e4756aedc2eb72914cb6aae1f676394d16dd4e124bedd4c356e133f9fe7",
          "master_fingerprint": "05c8b007",
          "path": "m/44'/1'/0'/0/8"
        }
      ]
    }
  ],
  "outputs": [
    {
    }
  ],
  "fee": 0.00005140
}
```
When I try to sign this, I get an error saying `Signature for index 0 already filled`

matejcik, 2019-05-06:
this signs fine for me in master. Note that I'm using master because released versions don't support regtest.

I'll try to reproduce on testnet somehow.

here's a JSON for trezorctl:
```json
{
    "inputs": [
        {
            "address_n": [2147483692, 2147483649, 2147483648, 0, 9],
            "prev_hash": "1d340c21cf93f442ce8d87ae615d0fd29de9634a5b40a5d9e4b2c82da6dadd94",
            "prev_index": 1,
            "sequence": 4294967295,
            "input_type": "SPENDADDRESS"
        },
        {
            "address_n": [2147483692, 2147483649, 2147483648, 0, 8],
            "prev_hash": "1d340c21cf93f442ce8d87ae615d0fd29de9634a5b40a5d9e4b2c82da6dadd94",
            "prev_index": 2,
            "amount": 100000000,
            "sequence": 4294967295,
            "input_type": "SPENDWITNESS"
        }
    ],
    "outputs": [
        {
            "amount": 199994860,
            "address": "bcrt1qq2kdfuapcmmrtqla8gv0wj854kujma054d4d33",
            "output_type": "PAYTOADDRESS"
        }
    ],
    "details": {},
    "prev_txes": {
        "1d340c21cf93f442ce8d87ae615d0fd29de9634a5b40a5d9e4b2c82da6dadd94": {
            "version": 2,
            "lock_time": 0,
            "inputs": [
                { 
                    "prev_hash": "de1c178a0edb8d42bbfbcedbaf5330b91d89248273eebb982024081bc0aa7ccd",
                    "prev_index": 1,
                    "script_sig": "",
                    "sequence": 4294967294
                }
            ],
            "bin_outputs": [
                {
                    "amount": 3099982120,
                    "script_pubkey": "00147b05ee92156374ac251c99fb5906f575fb9602c9"
                },
                {
                    "amount": 100000000,
                    "script_pubkey": "76a9149e3940a51dd2ff0753498229f921d2789ef3533088ac"
                },
                {
                    "amount": 100000000,
                    "script_pubkey": "0014fc534bfaacf6fd81cdec090769716663ebae5e58"
                }
            ]
        }
    },
    "coin_name": "Regtest"
}
```

matejcik, 2019-05-06:
no, my bad, i'm passing the wrong fields

matejcik, 2019-05-06:
managed to reproduce the problem.
here's a trezorctl reproducer JSON:
```json
{
    "inputs": [
        {
            "address_n": [2147483692, 2147483649, 2147483648, 0, 9],
            "prev_hash": "1d340c21cf93f442ce8d87ae615d0fd29de9634a5b40a5d9e4b2c82da6dadd94",
            "prev_index": 1,
            "sequence": 4294967295,
            "script_type": 0
        },
        {
            "address_n": [2147483692, 2147483649, 2147483648, 0, 8],
            "prev_hash": "1d340c21cf93f442ce8d87ae615d0fd29de9634a5b40a5d9e4b2c82da6dadd94",
            "prev_index": 2,
            "amount": 100000000,
            "sequence": 4294967295,
            "script_type": 3
        }
    ],
    "outputs": [
        {
            "amount": 199994860,
            "address": "tb1qnspxpr2xj9s2jt6qlhuvdnxw6q55jvygcf89r2",
            "script_type": 0
        }
    ],
    "details": {},
    "prev_txes": {
        "1d340c21cf93f442ce8d87ae615d0fd29de9634a5b40a5d9e4b2c82da6dadd94": {
            "version": 2,
            "lock_time": 0,
            "inputs": [
                { 
                    "prev_hash": "de1c178a0edb8d42bbfbcedbaf5330b91d89248273eebb982024081bc0aa7ccd",
                    "prev_index": 1,
                    "script_sig": "",
                    "sequence": 4294967294
                }
            ],
            "bin_outputs": [
                {
                    "amount": 3099982120,
                    "script_pubkey": "00147b05ee92156374ac251c99fb5906f575fb9602c9"
                },
                {
                    "amount": 100000000,
                    "script_pubkey": "76a9149e3940a51dd2ff0753498229f921d2789ef3533088ac"
                },
                {
                    "amount": 100000000,
                    "script_pubkey": "0014fc534bfaacf6fd81cdec090769716663ebae5e58"
                }
            ]
        }
    },
    "coin_name": "Testnet"
}
```

matejcik, 2019-05-06:
hm, there's not one but several bugs in TT's signing, seemingly because it was written with the assumption that outputs don't mix.

i can fix this case, but proper resolution depends on #57

closed 2019-05-07 by matejcik

achow101, 2019-12-17:
It seems like this wasn't fully resolved. I'm getting a `Some signatures are missing!` error now.

matejcik, 2020-04-22:
a full fix will be available through #617 which is now in active development

commit bf067cc0 (2019-05-07): core: better handling of TxRequest with mixed inputs

fixes #133