trezor/trezor-firmware #136: EthereumAddress message changed

ligi, 2019-05-06:
it seems the protobuf message for EthereumAddress has changed:

new:

https://github.com/trezor/trezor-firmware/blob/master/common/protob/messages-ethereum.proto#L46

old:

https://github.com/trezor/trezor-common/blob/dd8ec3231fb5f7992360aff9bdfe30bb58130f4b/protob/messages.proto#L337

this is quite bad as this makes it really ugly to support TREZOR devices with old and new firmware. I kind of tend to now drop support for the old firmware. But it would be great to avoid such things in the future. Ideally by not changing an existing message - but when needing to change introduce a new message.

matejcik, 2019-05-07:
the protobuf message itself is actually backwards-compatible; you see the bytes field has id 1 and the string field has id 2.
We should have left the old field in the definition, then you would be able to test and use whichever is set. You can still do that, but you have to hack the proto definition :/

perhaps it's worth returning the field?

ligi, 2019-05-07:
> perhaps it's worth returning the field?

that would actually be great!

btw. what is the best way to find the protobuf definitions for firmware 2.1.0? As far as I see for that version it still links to trezor-common - but there are no tags in there.

matejcik, 2019-05-07:
the only way is to checkout tag `core/2.1.0` and get the submodule commit from there

ligi, 2019-05-07:
ah - it's a submodule - that makes sense

ligi, 2019-05-07:
and PS: by all the criticism I had over the recent changes - really like how you split the proto files so I can only pick the ethereum relevant ones ..

commit d378f05d (2019-05-07): protob: add deprecated field for old-style EthereumAddress

this allows 3rd parties to keep backwards compatibility for Trezors
older than 1.8.0/2.1.0

fixes #136
closed 2019-05-07 by matejcik