trezor/trezor-firmware #144: Trezor T freezes after holding a button for a long time

matejcik, 2019-05-09:
seems related to #22 and also mentioned in #90 

fw version is 2.1.0

it looks a lot like return of the dreaded touch bug :(
https://github.com/trezor/trezor-core/issues/334

to reproduce:
1. on homescreen, hold down your finger for ~1 minute
2. do anything

expected:
- you can use trezor

actual:
- trezor doesn't react to touch anymore, until disconnect/reconnect

The problem is probabilistic and sometimes manifests after less than 10s (so in theory it could trigger when someone just presses a button a little longer). Holding for 1 minute seems to trigger it in 9 out of 10 tries.

Interestingly, I didn't manage to reproduce the problem on PIN input screen.

sancoder, 2019-05-21:
I think I found the root cause of the problem by just reading the code.
Here
https://github.com/trezor/trezor-firmware/blob/master/core/src/trezor/ui/mnemonic.py
we have edit_loop which waits for io.TOUCH
What if the event is not TOUCH but TOUCH_END?
It will wait for TOUCH but current event is TOUCH_END which is never gets dispatched.
The obvious fix is to check for event after timeout.

prusnak, 2019-05-21:
@matejcik @jpochyla can you please check and confirm?

jpochyla, 2019-05-22:
Those are two unrealated types. `TOUCH` is an interface, `TOUCH_START`, `TOUCH_START`, `TOUCH_MOVE` and `TOUCH_END` are events.

bosomt, 2019-05-27:
We have same problem.
Device is asking you to confirm anything on display and you close wallet then device freezes till you send another request.

Is it possible to implement some kind of timeout of dialogue So it fill reflect that computer was locked, browser was closed or connection time-outed after long time and there is no  reply from browser ?

matejcik, 2019-05-27:
> We have same problem.
> Device is asking you to confirm anything on display and you close wallet then device freezes till you send another request.

please let's not mix issues

read the reproducer instructions, notice that this is about device freezing until disconnected, regardless of USB communication. your issue OTOH is apparently about device freezing until a new request arrives

pankrats, 2019-05-29:
It happens basically everywhere. It is enough to hold your finger on the screen for ~30 seconds. 

Create backup (onboarding):
[image]

Set up PIN lock (onboarding)/ enable PIN in Basic settings:
[image]

Show on Trezor:
[image]

I can not reproduce it in "Enter your PIN" dialog when unlocking the device.

sorooris, 2019-05-31:
Simulated the bug when wrong PIN was entered on the device for the 15th time (16384s timeout) without touching the display, device froze at about 14000s mark.

andrewkozlik, 2019-06-25:
I think this issue was fixed by c1f54329049369bf7e102f45d1777ed941cb05a2 and 1ec5d091fde60ed17ff51661c319ce99dab9c379. @sorooris can you please test this with the latest firmware (2019-07)?

bosomt, 2019-06-25:
i was not able to reproduce it with 2.1.2 fw

pankrats, 2019-06-25:
Me neither, it does not happen with 2.1.2 ( core-2019-07-1ec5d0.bin), let's close the issue.

closed 2019-06-25 by pankrats