trezor/trezor-firmware #162: SD card filesystem

prusnak, 2019-05-20:
In order to use SD card, we need to decide which filesystem will be used.

Currently I see two options:
* FAT32 - https://github.com/micropython/oofatfs
* LittleFS - https://github.com/ARMmbed/littlefs

FAT32 | LittleFS
----|----
:heavy_plus_sign: major OS support | :heavy_minus_sign: not supported on major OS
:heavy_minus_sign: not power-loss resilient | :heavy_plus_sign: power-loss resilience
:heavy_minus_sign: not optimized for SD cards | :heavy_plus_sign: dynamic wear leveling

ciny, 2019-05-21:
WSL2 should add full linux system call compatibility to windows 10 so FUSE should work [source](https://devblogs.microsoft.com/commandline/announcing-wsl-2/), hopefully [littlefs-fuse](https://github.com/ARMmbed/littlefs-fuse) will too. It won't be "out of the box" support but better than nothing.

tuxxy, 2019-06-03:
I'm +1 on FAT32 because of the OS support.
Are there are other goals of including the SD card support that may sway me towards a different FS?

prusnak, 2019-06-26:
Implemented in https://github.com/trezor/trezor-firmware/pull/264

closed 2019-06-26 by prusnak