trezor/trezor-firmware #164: Trezor T Emulator Crashes

achow101, 2019-05-21:
I'm writing automated tests for HWI for the Trezor T and I've been running into some issues with the emulator crashing. There are two different errors I have seen. 

```
FATAL ERROR:
expr: sectrue * (0 == bind(usb_ifaces[i].sock, (struct sockaddr *)&usb_ifaces[i].si_me, sizeof(struct sockaddr_in)))
file: embed/unix/usb.c:101
func: usb_start
rev : a997ee5f

Please contact TREZOR support.
```

This error appears to be related to the socket used for the udp interface. It looks like it happens only during the emulator shutdown at the end of each test case so that's not really a concern.

```
FATAL ERROR:
expr: flash_write_word(norcow_sectors[sector], offset, prefix)
file: vendor/trezor-storage/norcow.c:85
func: norcow_write
rev : a997ee5f

Please contact TREZOR support.
```

This error is what's killing my tests. I am unsure of what is going on here.

prusnak, 2019-05-22:
The first error is just telling you that you have another instance of emulator running and it cannot bind the listening port. Don't you start another instance of the emulator before the previous one has properly closed the listening socket?

The second error is probably related. Maybe the new instance of the emulator has overwritten the storage of the old instance.

achow101, 2019-05-22:
> The first error is just telling you that you have another instance of emulator running and it cannot bind the listening port. Don't you start another instance of the emulator before the previous one has properly closed the listening socket?

> The second error is probably related. Maybe the new instance of the emulator has overwritten the storage of the old instance.

Indeed, the issue was the multiple emulators were running at the same time due to child processes.

closed 2019-05-22 by achow101