trezor/trezor-firmware #168: EOS signatures not compatible with CLEOS

cryptosig, 2019-05-22:
Trezor produces signatures in RSV format, but CLEOS, and probably every other EOS wallet, expects a signature that is already on the curve. Additionally the output signature should start with "SIG_K1". Whilst converting the rsv signature scheme in a transaction to a compatible EOS signature is not insurmountable, is there any reason a user should do further conversions of the output to sign the transactions in an expected EOS signature format?

tsusanka, 2019-05-23:
Let's ask @atkachyshyn who is the author of the EOS app

matejcik, 2019-05-23:
the RSV format was in original design because it was copied over from Ethereum, and we kept it to remain compatible with Keepkey's implementation which is using our protobufs.

however AFAICT keepkey has not released EOS support at this point? @keepkeyjon any comments on this?

so i'm in favor of switching the output format, preferably before this is released

not sure about SIG_K1. i know very little about EOS's interchange formats, I was under the impression that the signature that's actually used is simply concatenation of V+R+S?

in any case we can do all this in python-trezor

matejcik, 2019-05-23:
to clarify: I think Trezor should not return RSV, but the 65-byte binary blob.
then in python-trezor we can convert to b58 and prefix with "SIG_K1_" for text output.

but you could also argue that the string representation should come out of Trezor directly

keepkeyjon, 2019-05-23:
> however AFAICT keepkey has not released EOS support at this point?

It's in the firmware, but we haven't released a frontend for it yet. I'm happy to change the format, even if it breaks compatibility on those protobuf messages.... I don't expect anyone is using it yet.

keepkeyjon, 2019-05-23:
That also reminds me... we ought to try to reach consensus on SLIP48 with EOS. Is your plan to match the Ledger implementation and do `m/44'/194'/account'/0/0` path for owner and active keys, or do you want to put those two keys at `m/48'/4'/0'/account'/key'` and `m/48'/4'/1'/account'/key'` respectively? We have the flexibility to change that now while we don't have lots of users on it, and it'll be a lot harder to change later.

matejcik, 2019-05-23:
Is there anyone who actually _uses_ SLIP-48? I didn't think it existed in practice.

In any case we're staying with the SLIP44-based path for the release, but we can easily support both. But this is more of a question for wallet developers. We don't intend to provide first-party EOS wallet in the foreseeable future. And there's Fairy Wallet with Trezor integration, which presumably uses Ledger-like paths. If you intend to have a wallet that uses SLIP48, let's talk.

> > however AFAICT keepkey has not released EOS support at this point?
> 
> It's in the firmware, but we haven't released a frontend for it yet. I'm happy to change the format, even if it breaks compatibility on those protobuf messages.... I don't expect anyone is using it yet.

any opinion on bytes blob vs str representation of the signature?

cryptosig, 2019-05-23:
IMO any signature that outputs from Trezor should be able to be broadcasted without the need for further manipulation. I have found that [this](https://eosio.stackexchange.com/questions/3642/how-does-eos-create-signature) that might assist you

> A transaction signature is formed of a SIG_ prefix, a part that defines which curve algorithm was used to create the signature e.g; K1 or R1, and the base58 encoded SHA256 hash. e.g; SIG_K1_{base58_hash}

> 
It has some code at the end that might be useful to recreating the signatures without recreating the wheel.

matejcik, 2019-05-23:
> IMO any signature that outputs from Trezor should be able to be broadcasted without the need for further manipulation.

Does EOS allow you to broadcast bare signatures? Trezor doesn't give you back the rest of the tx anyway.

Or, a better question: how exactly do you use the output?

cryptosig, 2019-05-24:
EOS, like other Graphene networks, uses JSON in it's transactions. In order to have Trezor work, you output a transaction in JSON, the signature JSON array is blank. You then trezorctl sign the transaction json, the outputted signature should be able to be dropped back into the signature JSON array. ie
```
cleos push action -s -d -j contract action {} -p user@active
```
BEFORE signature(raw transaction from CLEOS):
```
{
  "expiration": "2019-05-22T05:29:12",
  "ref_block_num": 20470,
  "ref_block_prefix": 221063398,
  "max_net_usage_words": 0,
  "max_cpu_usage_ms": 0,
  "delay_sec": 0,
  "context_free_actions": [],
  "actions": [{ ... }],
  "transaction_extensions": [],
  "signatures": [],
  "context_free_data": []
}
```
AFTER signature:
```
{
  "expiration": "2019-05-22T05:29:12",
  "ref_block_num": 20470,
  "ref_block_prefix": 221063398,
  "max_net_usage_words": 0,
  "max_cpu_usage_ms": 0,
  "delay_sec": 0,
  "context_free_actions": [],
  "actions": [{ ... }],
  "transaction_extensions": [],
  "signatures": [SIG_K1_{base58_hash}],
  "context_free_data": []
}
```
then push the transaction JSON file using CLEOS
```
cleos push transaction "$(cat transaction.json)"
```

prusnak, 2019-05-24:
I agree that the `EosSignedTx` should look like this if that is what the whole EOS ecosystem is doing:

```
message EosSignedTx {
    optional string signature = 1; // Computed signature
}
```

I am rewriting the code and unit tests to reflect this change.

prusnak, 2019-05-24:
@cryptosig do you have a tool where I could convert RSV to SIG_K1 signature format so I can check I did it correctly?

cryptosig, 2019-05-24:
I don't have a tool, however have a look at the js:
https://github.com/EOSIO/eosjs-ecc/blob/master/src/signature.js

this is a Signature class that accepts r,s and than has conversion to the proper sig_ output

prusnak, 2019-05-24:
Implemented in 1583ea62f9f6fe83091f02d44626cc470b4d81af

closed 2019-05-24 by prusnak