trezor/trezor-firmware #172: EOS signatures don't match Public Key

cryptosig, 2019-05-24:
When I get an EOS public key from trezorctl and attach that value to an account, then try to perform a transaction the signature key doesn't match the public key given.
```
./trezorctl eos-get-public-key -n "m/44'/194'/0'/0/0"
WIF: EOS5PHaHGMyioSzWCjY84PFiZ6Mnn6KzY5by5G5v42AZeiaKzHzd8
Raw: 0241630cd74930361f2b90a07109ea614f0649c07ef52206891119cddc1ae4f5ff
./trezorctl eos-sign-transaction -n "m/44'/194'/0'/0/0" --file test6.json
Please confirm action on your Trezor device
EosSignedTx (103 bytes) {
    signature: 'SIG_K1_KdpCo21vwxXvhFTwCdcZTnP7tWr3oHD3U8WiXyzkPVn2XkaSmyELwMmjmUeYwk7vjTLjuxqicx4EeUHf5kV65AJUKCEg3i',
}
```
From CLEOS:
```
cleos push transaction -s "$(cat test6.json)"
Error 3090003: Provided keys, permissions, and delays do not satisfy declared authorizations
Ensure that you have the related private keys inside your wallet and your wallet is unlocked.
Error Details:
transaction declares authority '{"actor":"miniminimini","permission":"active"}', but does not have signatures for it under a provided delay of 0 ms, provided permissions [], provided keys ["EOS8UQGUQvfNhz9HJrmCq2PrhoLnT6dfQAeWksuyRChTDLBgLLLJX"], and a delay max limit of 3888000000 ms
```

cryptosig, 2019-05-24:
Update: setting the transaction delay to a value other than 0 seems to change the key, but the signature still isn't accepted. Maybe the signature is on the wrong curve?
```
transaction declares authority '{"actor":"miniminimini","permission":"active"}', but does not have signatures for it under a provided delay of 3888000000 ms, provided permissions [], provided keys ["EOS65fsLc1uxU4yqeqQfspGFSb4FbU1P5XqAYxoaXNwtUQokiE97X"], and a delay max limit of 3888000000 ms
```

prusnak, 2019-05-24:
What happens if you use the key `EOS_65fsLc1uxU4yqeqQfspGFSb4FbU1P5XqAYxoaXNwtUQokiE97X` i.e. extra underscore after EOS?

prusnak, 2019-05-24:
I implemented new pubkey format in master (a6e51434f25401d4caf982dfcae341a957dbc58f). Can you please try if this works?

cryptosig, 2019-05-24:
I get an invalid key when trying to use the new key, only the last bit changed at the end of the string. 
```
old  =EOS5PHaHGMyioSzWCjY84PFiZ6Mnn6KzY5by5G5v42AZeiaKzHzd8
new=EOS5PHaHGMyioSzWCjY84PFiZ6Mnn6KzY5by5G5v42AZeiaGtLWVd
```
Did you add the checksum to your compressed publickey? When I look at the code in apps/eos/get_public_key.py I didn't see a checksum concatenated to the compressed public key.
IE 
checksum = ripe160(compressed public key).head8bytes
address=base58(compressed public key+checksum) 
eosaddress="EOS"+address

prusnak, 2019-05-24:
It is added. Checksum however changes when you add prefix. This means you cannot just change/replace the prefix.

You might want to invite some of the EOS developers into this discussion to help debugging.

cryptosig, 2019-05-24:
This is not correct then, the public keys on EOS and all graphene networks are interchangeable. such that:
```
Baseaddress:78GCKGX3C4Tr8iVQamZ4UVTgEo84RQhQaHndGB63YFqAx68LMd
EOSaddress:EOS78GCKGX3C4Tr8iVQamZ4UVTgEo84RQhQaHndGB63YFqAx68LMd
BTSaddress:BTS78GCKGX3C4Tr8iVQamZ4UVTgEo84RQhQaHndGB63YFqAx68LMd
PPYaddress:PPY78GCKGX3C4Tr8iVQamZ4UVTgEo84RQhQaHndGB63YFqAx68LMd
STEEMaddress:STEEM78GCKGX3C4Tr8iVQamZ4UVTgEo84RQhQaHndGB63YFqAx68LMd
```

prusnak, 2019-05-25:
@cryptosig you are still using legacy format, not the one that starts with `PUB_K1_`. Get a cleos developer into this thread so we can discuss.

cryptosig, 2019-05-25:
I have been dissecting and debugging the trezor code. I have taken the private key out of the emu and have been using it to do transaction tests. The old public address, before the changes(a6e5143), the one that was EOS5PHaHGMyioSzWCjY84PFiZ6Mnn6KzY5by5G5v42AZeiaKzHzd8, was the correct public address key, only the private signature wasn't working. So I will be rolling back the git to test out the signature differences between the CLEOS signature and the Trezor signature to figure out where it's going wrong. In the mean time hopefully we can get some devs in here.

prusnak, 2019-05-26:
OK, I reverted the new pubkey format in 07ce9357253e9c7bd6b0f3245756a9e478d5c38f

Let's see what EOS devs come up with.

matejcik, 2019-05-30:
@cryptosig 	your original error suggests that you're signing a transaction for an account for which you don't have permissions o_O
i'm playing with jungle testnet and can sign my own transactions fine.

cryptosig, 2019-05-30:
I have achieved a valid signature and transfer of EOS with the latest code.
Steps taken:
1) build the transaction with a long expiration, to give yourself enough time to gather signatures
```
cleos push action -s -d -j -x 300 eosio.token transfer '{"from": "miniminimini", "to": "maximaximaxi", "quantity": "1 EOS", "memo":"testtest"}' -p miniminimini@active

{
  "expiration": "2019-05-30T12:26:23",
  "ref_block_num": 13000,
  "ref_block_prefix": 1953856974,
  "max_net_usage_words": 0,
  "max_cpu_usage_ms": 0,
  "delay_sec": 0,
  "context_free_actions": [],
  "actions": [{
      "account": "eosio.token",
      "name": "transfer",
      "authorization": [{
          "actor": "miniminimini",
          "permission": "active"
        }
      ],
      "data": "e0a6936e3ae9a693e0ba91ae1be9ba91010000000000000000454f5300000000087465737474657374"
    }
  ],
  "transaction_extensions": [],
  "signatures": [],
  "context_free_data": []
}
```
2) Convert data to json for trezor, and replace data string with  data from {...}
```
cleos convert unpack_action_data eosio.token transfer e0a6936e3ae9a693e0ba91ae1be9ba91010000000000000000454f5300000000087465737474657374
{
  "from": "miniminimini",
  "to": "maximaximaxi",
  "quantity": "1 EOS",
  "memo": "testtest"
}
```
3) sign the transaction and copy the signature into the signature:[] of the json transaction
```
./trezorctl eos-sign-transaction -n "m/44'/194'/0'/0/0" -f ~/test.json

Please confirm action on your Trezor device
EosSignedTx (103 bytes) {
    signature: 'SIG_K1_KbkF97aBdxEYiw4hb2FQbZ7PT7WfBe69bV78y7TkLGwdXzLrUURBCraRG4aSeYMFhNGvqM3uXwv9xUai1Y8VHj1pZTAfp7',
}
```

4) broadcast the transaction, output should look like below
```
cleos push transaction -s "$(cat test10.json)"
{
  "transaction_id": "c819e3937dfb3b78aa9c3ae79ff57015125b44ecd107d734f35770efc056bb53",
  "processed": {
    "id": "c819e3937dfb3b78aa9c3ae79ff57015125b44ecd107d734f35770efc056bb53",
    "block_num": 1651602,
    "block_time": "2019-05-30T12:23:04.000",
    "producer_block_id": null,
    "receipt": {
      "status": "executed",
      "cpu_usage_us": 1831,
      "net_usage_words": 17
    },
    "elapsed": 1831,
    "net_usage": 136,
    "scheduled": false,
    "action_traces": [{
        "receipt": {
[... 109 lines cut ...]
                "quantity": "1 EOS",
                "memo": "testtest"
              },
              "hex_data": "e0a6936e3ae9a693e0ba91ae1be9ba91010000000000000000454f5300000000087465737474657374"
            },
            "context_free": false,
            "elapsed": 24,
            "console": "",
            "trx_id": "c819e3937dfb3b78aa9c3ae79ff57015125b44ecd107d734f35770efc056bb53",
            "block_num": 1651602,
            "block_time": "2019-05-30T12:23:04.000",
            "producer_block_id": null,
            "account_ram_deltas": [],
            "except": null,
            "inline_traces": []
          }
        ]
      }
    ],
    "except": null
  }
}
```

FYI, there might be still a bounty for getting EOS working on Trezor. Contact the BP Cypherglass to inquire if it is still available.

closed 2019-05-30 by cryptosig