trezor/trezor-firmware #174: Investigate CCRAM usage

prusnak, 2019-05-24:
Our older experiments introduced usage of CCRAM in the firmware.

See https://github.com/trezor/trezor-firmware/commit/265e23b2aaa97091dfacfc79e79287035bd9b2eb

However, there are some outstanding issues with that:

See https://github.com/trezor/trezor-firmware/issues/66#issuecomment-473247788

romanz, 2019-12-12:
Just wanted to ask whether we may use https://github.com/romanz/trezor-firmware/commit/fa5ec1cde6aebdb9930771056c62bf041de3d260 (since moving BSS to CCMRAM seems to work) in order to increase the available stack for https://github.com/trezor/trezor-firmware/pull/317?

prusnak, 2019-12-15:
@romanz Yes, this seems like a reasonable workaround.

romanz, 2020-01-18:
There seems to be a simpler way to implement range-proof & surjection proofs:
https://github.com/trezor/trezor-firmware/pull/317#issuecomment-575892242

TychoVrahe, 2025-01-23:
we use ccmram on F4 models now

closed 2025-01-23 by TychoVrahe