trezor/trezor-firmware #178: KMD tests fail for T1 and T2

prusnak, 2019-05-25:
`python/trezorlib/tests/device_tests/test_msg_signtx_komodo.py`

prusnak, 2019-05-26:
T1 error:

```
E         - 0400008085202f89019ca1ad57524f6076d8d296c17bdefbb9fc162072fac3eb659f220e61783e9c33000000006a47304402205055d1af7ace3e01920d3a4594eaa095eb63fa3d7c2ec135cccd8dd50b1d30580220487a1a06031fabfeaafc24554d79220166eeb5a3f377586f0f14b8ca0d19de4a012102a87aef7b1a8f676e452d6240767699719cd58b0261c822472c25df146938bca5ffffffff01e054e111000000001976a91433058d6bb20e9297fc0a518e2e0262e854496a6c88ac6be15c5c000000000000000000000000000000
E         + 0400008085202f89019ca1ad57524f6076d8d296c17bdefbb9fc162072fac3eb659f220e61783e9c33000000006a47304402206972af8ff4dec4074da9edb1a741114bebda9e686bb411dd6f388aafd81f0af2022060a05d0ea66d5b8632868b6a6d377ad1a8941385d628be4631b31cc246014b8501210235ad92bb4efda1e6794890f248fa26aab75906bd496c07a6a8532b62a5bd80f7ffffffff01e054e111000000001976a91433058d6bb20e9297fc0a518e2e0262e854496a6c88ac6be15c5c000000000000000000000000000000
```

TT error:

```
E         - 47304402205055d1af7ace3e01920d3a4594eaa095eb63fa3d7c2ec135cccd8dd50b1d30580220487a1a06031fabfeaafc24554d79220166eeb5a3f377586f0f14b8ca0d19de4a012102a87aef7b1a8f676e452d6240767699719cd58b0261c822472c25df146938bca5ffffffff0
E         + 47304402206972af8ff4dec4074da9edb1a741114bebda9e686bb411dd6f388aafd81f0af2022060a05d0ea66d5b8632868b6a6d377ad1a8941385d628be4631b31cc246014b8501210235ad92bb4efda1e6794890f248fa26aab75906bd496c07a6a8532b62a5bd80f7ffffffff...
E         
```

ca333, 2019-05-26:
Thanks for the notification @prusnak - i did check it again some time ago and noticed that we are also missing: https://github.com/ca333/trezor-mcu/commit/8f879f7160dd50d4b1e6d9eb69bd6ed81c5d83f7#diff-fb20d8a8841a1cea41e6179f2c8485bdR693 in order to use KMD and the special "reward transactions" since the output will be bigger than input. Will check the signing code in the coming days and get back to this.

ZdenekSL, 2019-06-26:
@ca333 Any news on this? Thank you

ca333, 2019-07-07:
@ZdenekSL we are back on it and looking into the test-issues.

ca333, 2019-08-08:
@ZdenekSL @prusnak all issues are resolved with https://github.com/trezor/trezor-firmware/pull/354 - thank you

prusnak, 2019-08-09:
Fixed via e23bb10ec49710cc2b2b993db9c907d3c7becf2c

closed 2019-08-09 by prusnak