trezor/trezor-firmware #184: Add upgrade tests to CI

tsusanka, 2019-05-28:
Add tests to CI, which will try upgrades from older versions.

@prusnak already started this using gitlab/trezor-public/artifacts

Let's add this to CI. We might need to migrate from gitlab/trezor-public/artifacts to the dev website introduced by https://github.com/trezor/trezor-firmware/issues/183

tsusanka, 2019-05-28:
@prusnak I'm not sure how to deal with the invalid link of `ld-linux-x86-64.so.2`. Should I rebuild? I know you've mentioned you've applied some patch

```
/nix/store/681354n3k44r8z90m35hm8945vsp95h1-glibc-2.27/lib/ => /usr/lib64/ld-linux-x86-64.so.2 (0x00007f5174197000)
```

prusnak, 2019-05-28:
This is fixable via patchelf, @1000101 or @mmahut  might help. We need to change the rpath of the interpreter stored in the binary.

Another way would be to rebuild all of course, but it is kind of tedious job to do.

mmahut, 2019-05-28:
@tsusanka I don't have access to the binaries in question, but you can try to run ``autoPatchelfFile`` against the binary.

```
$ nix-shell -p autoPatchelfHook

[nix-shell:~]# autoPatchelfFile $(which hello)
searching for dependencies of /nix/store/gdh8165b7rg4y53v64chjys7mbbw89f9-hello-2.10/bin/hello

[nix-shell:~]# 
```

1000101, 2019-05-28:
> This is fixable via [patchelf](https://nixos.org/patchelf.html), @1000101 or @mmahut might help. We need to change the rpath of the interpreter stored in the binary.
> 
> Another way would be to rebuild all of course, but it is kind of tedious job to do.

Fixed

prusnak, 2019-05-28:
> @tsusanka I don't have access to the binaries in question, but you can try to run `autoPatchelfFile` against the binary.

@mmahut the problem was the opposite. the binary was produced on NixOS and we need to run it on Debian - that is we needed to remove fancy NixOS rpaths with empty ones. :)

tsusanka, 2019-05-29:
Duplicate of #126

closed 2019-05-29 by tsusanka