trezor/trezor-firmware #187: Dry-run recovery

bosomt, 2019-05-28:
# Steps to reproduce issue 

https://blog.trezor.io/test-your-seed-backup-dry-run-recovery-df9f2e9889

# Expected result 
 If the seeds match, the TREZOR Wallet will tell you your backup is okay. If it is not the same, then the Wallet will show you an error.

# Actual result 
freezing issue

# Other information 
- Reported by: Daria Roik 

- Test case number: TC03 

- Test run: [https://satoshilabs.testlodge.com/projects/32891/runs/399820?tab=2&run_section_id=582320#executed_case_17250491](https://satoshilabs.testlodge.com/projects/32891/runs/399820?tab=2&run_section_id=582320#executed_case_17250491)

bosomt, 2019-05-28:
duplicity

pankrats, 2019-05-28:
Sorry, false alarm

closed 2019-05-28 by bosomt

bosomt, 2019-06-03:
An associated test has been re-run in TestLodge. 

**Result:** Skipped 

**Other information** 

- Update by: Sebastian Soroori 

- Test case number: TC03 

- Test run: https://satoshilabs.testlodge.com/projects/32891/runs/400303?tab=2&run_section_id=582968#executed_case_17272371